Ultimaker/Cura #9068: Dual extruder preview mode rendered wrong

sundansx, 2021-01-04:
**Application version**
4.8.0

**Platform**
Windows 10 64bit, build 2004, Nvidia 1080ti with 456.71 drivers

**Printer**
Qidi Tech 1, custom profile.

**Reproduction steps**
1. set models to second extruder, and change supports and skirt to the same extruder
2. slice then preview.  preview is rendered with extruder one color.

**Screenshot(s)**
see video.  
https://user-images.githubusercontent.com/25075847/103496825-74b04480-4df4-11eb-9b83-0e9411be447c.mp4

**Actual results**
preview is rendered with wrong extruder color

**Expected results**
preview is rendered with correct extruder color

**Project file**
[previewBUG.zip](https://github.com/Ultimaker/Cura/files/5762798/previewBUG.zip)

**Log file**
I don't see anything in the log related to this problem.

**Additional information**
Cura 3.5.1 works as expected with this custom configuration (had to tweak the config files slightly for 4.8).

fvrmr, 2021-01-12:
I reproduced this, but I coudn't reproduce your issue. Maybe you have another project file with the same issue?
[image]

sundansx, 2021-01-12:
I reloaded the .3mf into my 4.8 and it still had the problem.  **Do you think it is realted to my printer definitions**?  I have included them in case someone has some time to look at them.  They are based off the ones linked above and the Ultimaker original dual profiles for printer and extruders.  Other than those, my setup is pretty stock and a clean install (did not import 3.5 settings).
[cura.zip](https://github.com/Ultimaker/Cura/files/5805063/cura.zip)

Ghostkeeper, 2021-01-13:
Yes, the problem is in those printer definitions. Particularly the "left" one (the second one, which as black material in it in your project). This is the definition for that extruder:

```
{
    "version": 2,
    "name": "Left Extruder",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "qiditech_i",
        "position": "1"
    },

    "overrides": {
        "extruder_nr": { "default_value": 0 },
        "machine_nozzle_size": { "default_value": 0.4 },
        "material_diameter": { "default_value": 1.75 }
    }
}
```

Note that the position is 1 but the `extruder_nr` setting is set to 0. Set that to 1 and your problem is gone. This is a mistake in your mod, not in Cura.

closed 2021-01-13 by Ghostkeeper

sundansx, 2021-01-13:
oh snap!  Thank you, Ghost keeper, for looking through this and finding my mistake.  If I wanted to add this support for the QidiTech 1, would I do a pull request?  If I do, what are the chances that it would make it into master?

Ghostkeeper, 2021-01-25:
Yeah, a pull request is the way to go.

I'd say that chances that it would make it into master are really high, as long as the profiles load properly in Cura and the automated tests don't fail. Most contributed printers fail the automated tests the first time but it's always easy to fix.

We've got quite a backlog of pull requests to process but the easier ones for us (like a new printer definition) tend to go a bit faster. If you submit it now I think it'll almost certainly be in the next release.