Ultimaker/Cura #9070: Plugin Folder on ubuntu

vekexasia, 2021-01-04:
**Application version**
4.8.0

**Platform**
ubuntu 20.04.1

**Printer**
not relevant

**Reproduction steps**
1. install cura
2. try to find folder

**Actual results**
cant find the plugins folder. I tried ~/.local/share/cura and ~/.config...

**Expected results**
find some folder?

**Log file**
Cant find the log file as well

**Additional information**
I did try to install cura from snap before using apt...

fieldOfView, 2021-01-04:
Use the AppImage.

After running the AppImage for the first time, the settings can be found in ~/.local/share/cura and in ~/.config/cura

Ghostkeeper, 2021-01-13:
Could you try that, @vekexasia?

The plug-in folder should be located in `~/.local/share/cura/4.8/plugins`. Cura creates folders at `~/.local/share/cura`, `~/.config/cura` and `~/.cache/cura` for respectively its data (profiles and plug-ins), its preferences and its cache files. This is according to the XDG spec. If this is different when using Apt or Snap that is a bug in that distribution. We only maintain the AppImage here.

vekexasia, 2021-01-13:
Sorry, I tried and it works. Sorry for not reporting it back.

-- 

Andrea Baccega
[image: Email]me@andreabaccega.com
[image: Hangouts]vekexasia@gmail.com
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

closed 2021-01-14 by nallath

MaratZakirov, 2021-04-02:
@fieldOfView What is AppImage and why CURA does not work properly without it?

fieldOfView, 2021-04-02:
The AppImage is the distribution of Cura that is supported by Ultimaker. It can be downloaded here: https://ultimaker.com/software/ultimaker-cura