Ultimaker/Cura #9071: beginners

Nicki381, 2021-01-04:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

a123qwertz567, 2021-01-04:
This is the wrong place for this but no matter I suggest you the following channels. They helped me a lot. Beside this you just have to  try different settings again and again and again. Level your bed and check for other hardware malfunctions.

https://www.youtube.com/channel/UCbgBDBrwsikmtoLqtpc59Bw
https://www.youtube.com/user/beginnerelectronics

a123qwertz567, 2021-01-04:
And please shorten the heading of your issue that is an assault to my eyes....

jellespijker, 2021-01-04:
Hi @Nicki381 This is indeed the wrong forum for that. So I'm closing this issue.
But maybe you could check out the following articles for inspiration and help https://support.ultimaker.com/hc/en-us/sections/360003548619-Print-settings 

It might also be a good idea to install the user setting guide plugin. This plugin has extensive articles about the various Cura settings and can be downloaden from the marketplace in Cura.

closed 2021-01-04 by jellespijker

Nicki381, 2021-01-04:
Thank you. The trouble is when your only used to using word and understanding a whole new  3d settings its very overwhelming to start. Especially when your trying to understand how it all works. Ive got some fantastic quality small prints off cura once i starting to understand it. Thanks again for your help. And the cura software.