Ultimaker/Cura #9072: Slicing hangs with certain STL's on macOS

coatezy, 2021-01-04:
When slicing certain STL's on macOS, CuraEngine is crashing.

**Application version**
4.8.0

**Platform**
macOS Catalina 10.15.6

**Printer**
Creality Ender 3

**Reproduction steps**
- Add Ender-3_Display_Cover_factory_srews_with_bell_closed.STL which can be found here https://www.thingiverse.com/thing:2858209
- Hit Slice

**Actual results**
Slicing hangs

**Log file**
Process:               CuraEngine [22538]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine
Identifier:            CuraEngine
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        cura [393]
Responsible:           cura [393]
User ID:               501

Date/Time:             2021-01-04 20:35:32.129 +0000
OS Version:            Mac OS X 10.15.6 (19G2021)
Report Version:        12
Anonymous UUID:        4C425F56-C5F7-720C-CFA6-CCF0EC7E7307

Time Awake Since Boot: 3600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
dyld2 mode
abort() called
terminating with uncaught exception of type std::length_error: vector

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff68d8233a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff68e3ee60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff68d09808 abort + 120
3   libc++abi.dylib               	0x00007fff65f68458 abort_message + 231
4   libc++abi.dylib               	0x00007fff65f598a7 demangling_terminate_handler() + 238
5   libobjc.A.dylib               	0x00007fff67a945b1 _objc_terminate() + 104
6   libc++abi.dylib               	0x00007fff65f67887 std::__terminate(void (*)()) + 8
7   libc++abi.dylib               	0x00007fff65f6a1a2 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 27
8   libc++abi.dylib               	0x00007fff65f6a169 __cxa_throw + 113
9   libc++.1.dylib                	0x00007fff65f3cae8 std::__1::__throw_length_error(char const*) + 56
10  libc++.1.dylib                	0x00007fff65f44ed6 std::__1::__vector_base_common<true>::__throw_length_error() const + 16
11  CuraEngine                    	0x0000000104e7e1a6 cura::Infill::generateLinearBasedInfill(int, cura::Polygons&, int, cura::PointMatrix const&, cura::ZigzagConnectorProcessor&, bool, long long) + 1270
12  CuraEngine                    	0x0000000104e7ad55 cura::Infill::generateLineInfill(cura::Polygons&, int, double const&, long long) + 357
13  CuraEngine                    	0x0000000104e7951f cura::Infill::_generate(cura::Polygons&, cura::Polygons&, cura::SierpinskiFillProvider const*, cura::SliceMeshStorage const*) + 303
14  CuraEngine                    	0x0000000104e78bd9 cura::Infill::generate(cura::Polygons&, cura::Polygons&, cura::SierpinskiFillProvider const*, cura::SliceMeshStorage const*) + 457
15  CuraEngine                    	0x0000000104e56b66 cura::FffGcodeWriter::processSkinPrintFeature(cura::SliceDataStorage const&, cura::LayerPlan&, cura::SliceMeshStorage const&, unsigned long, cura::Polygons const&, cura::GCodePathConfig const&, cura::EFillMethod, cura::AngleDegrees, long long, cura::Ratio, cura::Polygons*, bool&, double) const + 486
16  CuraEngine                    	0x0000000104e56766 cura::FffGcodeWriter::processTopBottom(cura::SliceDataStorage const&, cura::LayerPlan&, cura::SliceMeshStorage const&, unsigned long, cura::PathConfigStorage::MeshPathConfigs const&, cura::SkinPart const&, cura::Polygons&, bool&) const + 2998
17  CuraEngine                    	0x0000000104e54ec3 cura::FffGcodeWriter::processSkinPart(cura::SliceDataStorage const&, cura::LayerPlan&, cura::SliceMeshStorage const&, unsigned long, cura::PathConfigStorage::MeshPathConfigs const&, cura::SkinPart const&) const + 195
18  CuraEngine                    	0x0000000104e50fef cura::FffGcodeWriter::processSkinAndPerimeterGaps(cura::SliceDataStorage const&, cura::LayerPlan&, cura::SliceMeshStorage const&, unsigned long, cura::PathConfigStorage::MeshPathConfigs const&, cura::SliceLayerPart const&) const + 1215
19  CuraEngine                    	0x0000000104e4d4a7 cura::FffGcodeWriter::addMeshPartToGCode(cura::SliceDataStorage const&, cura::SliceMeshStorage const&, unsigned long, cura::PathConfigStorage::MeshPathConfigs const&, cura::SliceLayerPart const&, cura::LayerPlan&) const + 615
20  CuraEngine                    	0x0000000104e4c586 cura::FffGcodeWriter::addMeshLayerToGCode(cura::SliceDataStorage const&, cura::SliceMeshStorage const&, unsigned long, cura::PathConfigStorage::MeshPathConfigs const&, cura::LayerPlan&) const + 966
[... 90 lines cut ...]
 
                                VIRTUAL   REGION 
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Kernel Alloc Once                    8K        1 
MALLOC                           149.0M       49 
MALLOC guard page                   16K        4 
STACK GUARD                       56.0M        2 
Stack                             8712K        2 
__DATA                             640K       41 
__DATA_CONST                        20K        1 
__LINKEDIT                       389.9M        3 
__OBJC_RO                         32.3M        1 
__OBJC_RW                         1908K        2 
__TEXT                            7260K       40 
shared memory                        8K        2 
===========                     =======  ======= 
TOTAL                            645.3M      148 

**Additional information**
Also tested on a PC running Ubuntu and exact same thing happens. Works in Windows but I am seeing some add defects in the print.

Also saw issue with another of the files (Ender-3_Display_Cover_with_bell) whereby the slice completes but the print time is estimated at 4 days and 17 hours when inspecting the preview I noticed this.

[image]

coatezy, 2021-01-04:
Another quick observation, the Ender-3_Display_Cover_factory_srews_with_bell_closed_3connectors STL file appears to slice fine on MacOS.

coatezy, 2021-01-04:
I can confirm Cura slices as expected when using version 4.7.1. No crashing on no 4 day prints. 😄

Ghostkeeper, 2021-01-14:
I'm not able to reproduce this crash, using Ubuntu. Not with the default settings for the Ender 3 anyway. It slices fine and takes 3.5 hours to print.
[image]

I haven't yet seen slicing crashes that are operating system specific.

Perhaps you could share a project file? There could be some things with the precise location of the object, or maybe any of the profiles were changed.

coatezy, 2021-01-14:
Thanks for looking in to this, @Ghostkeeper. I will run it again on Ubuntu and post the logs along with the project. There are others on Thingiverse that are also having the same problem and a thread under comment:

> Cura keeps getting stuck slicking this model. Any idea why?

One suggestion was to rotate the model 90 degrees. When I did this on the MacOS version using Ender-3_Display_Cover_factory_srews_with_bell_closed.STL it sliced but I ended up with the long line causing the print to take ~5 days.

coatezy, 2021-01-14:
Attached is two projects - one with the STL imported as is and the other rotated 90 degrees.

[Cura Project.zip](https://github.com/Ultimaker/Cura/files/5814570/Cura.Project.zip)

coatezy, 2021-01-14:
Updated zip with an extra project with the model rotated upwards 15 degrees at the front. It slices perfectly in this orientation with a print time of 3hours 51 (with no supports). Of course I wouldn't want to print in this way. 😆 

[Cura Project.zip](https://github.com/Ultimaker/Cura/files/5814831/Cura.Project.zip)

fvrmr, 2021-01-19:
Thanks for your project files, we can reproduce this with your files. So we are gonna look into this.

coatezy, 2021-01-19:
Glad the project files helped. 🙌  Let me know if you need anything else or if I can help in any way, testing etc.

jellespijker, 2021-01-19:
It fails on my computer (Arch Linux) for the 4.8 branch. When it is exporting the GCode it is terminated with a SigAbort
```shell
Starting export...
Multithreading GcodeLayerThreader with 1 threads.
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append
Signal: SIGABRT (Aborted)
```

Good thing is that it slices correctly on Master so it should be fixed (for this model at least) in the 4.9 release

I will have to dive into it a little deeper to see what the root cause is.

In the mean time you could try it out with our new experimental Alpha Arachne release, since this is based on Master after 4.8 was released. I personally have no problems slicing the model in that pre-release. See https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha

ne2pit, 2021-04-13:
Reproduced slicer crash on MacOS Sierra.
Tried every file in this thingiverse package. Only one sliced fine Ender-3_Display_Cover_factory_srews_with_bell_closed_3connectors.STL.

Used some online STL fixer. It does not find any critical errors. But made some corrections. Fixer log:

> -> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 14772ms ------
-> Vertex count changed from 13088 to 14185 (+1097)
-> Triangle count changed from 26196 to 28398 (+2202)

Here is original and fixed version of Ender-3_Display_Cover_factory_srews_with_bell_closed.STL: 
[Ender-3_Display_Cover_hang_and_fixed_stl.zip](https://github.com/Ultimaker/Cura/files/6306520/Ender-3_Display_Cover_hang_and_fixed_stl.zip)

Ghostkeeper, 2021-05-03:
@ne2pit Which version of Cura were you using?

ne2pit, 2021-05-05:
@Ghostkeeper I used MacOS Cura version 4.8.0. The same version as topic starter used.
Unfortunately I can not check 4.9.0 right now due to new MacOS version limit in Cura.

closed 2024-12-08 by github-actions[bot]