Ultimaker/Cura #9073: Incorrect slicing in 4.8.0 extrusion extends outside of surface

ArtieH, 2021-01-05:
**Application version**
4.8.0

**Platform**
Windows 10, Intel i5
**Printer**
Homebuilt Hypercube Evolution

**Reproduction steps**
1. Thingiverse user had issue slicing my model. (Cura estimated a very long print completion time.)
2. Verified model with Meshmixer which showed no model issues.
3. Sliced model with Cura 4.8.0 which estimated print time to be greater than 2 days.
4. Checked preview and found an extrusion line from layer 111 that extended outside the model to what seems infinity.
5. Changed the line width to 0.39mm from the default of 0.40mm and the model sliced correctly. (0.39mm is what I usually slice at.)
6. Re-sliced the model under Cura 4.7.1 with the line width at 0.40mm and the model sliced correctly.

**Cura 4.8.0 with line width of 0.40mm**
[image]

**Cura 4.8.0 with line width of 0.39mm**
[image]

**Cura 4.7.1 with line width of 0.40mm**
[image]

**Actual results**
Under 4.8.0 the slicing is not correct when line width is 0.40mm. It is correct when line width is 0.39mm.
Under 4.7.1 slicing is correct at 0.40mm and 0.39mm line width.

**Expected results**
Slicing should be correct under Cura 4.8.0

**Project file**
[Case_Rear_V2.zip](https://github.com/Ultimaker/Cura/files/5767230/Case_Rear_V2.zip)

(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
[cura_log.zip](https://github.com/Ultimaker/Cura/files/5767552/cura_log.zip)

(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**

Thingiverse listing
[https://www.thingiverse.com/thing:3852672](url)

Pertinent stl file 
[Case_Rear_V2_stl.zip](https://github.com/Ultimaker/Cura/files/5767568/Case_Rear_V2_stl.zip)

DarkFighterLuke, 2021-01-05:
I faced the same issue 2-3 weeks ago while printing an object on Ender 3.
I resolved with the 4.7.1 and then I am downgrading to that version.

konskarm, 2021-01-06:
Unfortunately we are unable to reproduce the issue with the given project file. Nevertheless, I suspect it is the same issue as the one posted in https://github.com/Ultimaker/Cura/issues/8736#issuecomment-725931474 where the slicing time increases significantly due to an infinite outer wall. This is already an issue in our backlog (devs see CURA-7846), so I will close this one in favor of #8736. 

In case you manage to reproduce your issue with a project file, please post it in the aforementioned thread.

closed 2021-01-06 by konskarm

konskarm, 2021-01-07:
Unfortunately I couldn't reproduce the issue with the given project file. It seems it does not generate the infinite wall at all. However, the same issue was reported in https://github.com/Ultimaker/Cura/issues/8736 with a reliably reproducible project file, so I will close this issue in favor of the other one. There is also a ticket already in our backlog spawned from that issue (devs see CURA-7846).
Please add any extra comments to that thread. Also, if you manage to have a 3mf file that reproduces it, please post it in #8736 so that we can use it when dealing with our internal ticket.