Ultimaker/Cura #9075: [Arachne_engine_alpha] Blobs in skin on thin walls.

saldot, 2021-01-05:
**Application version**
Arachne_Engine_Alpha 99.9.0

**Platform**
Windows 10, Nvidia GTX1080 TI

**Printer**
Ender 3

**Reproduction steps**
Slice a model that has thin wall that are slightly thicker than the nozzle size.
Use any Variable Line Strategy

**Screenshot(s)**
In 4.8 the outer shell is solid with a line wrapping all around the model leaving a smooth surface.
[image]

In Arachne alpha the shell is not a single line but interrupted leaving an uneven surface with blobs due to retractions.
[image]

**Actual results**
Uneven surfaces and poorer looking prints.

**Expected results**
The outer wall is, when possible, an uninterrupted single line.

**Project file**
[CE3PRO_Tensioner.zip](https://github.com/Ultimaker/Cura/files/5770317/CE3PRO_Tensioner.zip)

**Log file**
N/A

**Additional information**
N/A

Ghostkeeper, 2021-01-14:
This is currently expected behaviour.

Your case is basically the [Königsberg problem](https://en.wikipedia.org/wiki/Seven_Bridges_of_K%C3%B6nigsberg). There isn't always a solution. There is one Euler Path possible in your case, but it requires moving the seam to where it's not supposed to go. And there isn't always a solution. For instance, if the opposite side of that hexagon would be slightly thinner it would have a single line there too, resulting in 4 odd-edged vertices and thus no Euler Path.

We can consider loosening up the constraints of the seam placement. Then the problem remains how to create a minimal set of Euler paths. That is quite simple in a sparse graph (start from any odd-edged vertex), but our graph is fully connected with possible travel moves one could need to take in some cases which need to be the shortest travel moves. This is an NP-complete problem which we're just using Nearest Neighbour for, but that won't ensure a minimal set of Euler Paths of course.

In short, there's an amalgamation of graph algorithms governing this sort of situation, making it a complex problem to deal with. There are theoretical improvements possible, but they come at the cost of seam placement and it's unknown to me how hard those are.
For that reason I'll keep this issue open, but to temper expectations I need to defer it. We'll not be dealing with it before the stable release I think.

saldot, 2021-01-14:
The hole and hexagon are centered in the piece so both walls are equally thick on both sides.
I suspect the issue here is that the 2 shells share a wall that is just wide enough that it can contain both 1 or 2 lines and still be within the rules parameters i've set in the slicer - Minimum variable line width.

Would it be a good idea to use separate eulerian paths for shells even if they, as in this case, share a thin wall - if that wall is thick enough to contain 2 lines?
If the shells were separate eulerian paths the prints would only have 1 seem per shell instead of potentially 2 or 3 which would result in better looking prints.

Ghostkeeper, 2021-01-25:
> I suspect the issue here is that the 2 shells share a wall that is just wide enough that it can contain both 1 or 2 lines and still be within the rules parameters i've set in the slicer - Minimum variable line width.

It's certainly near the border, in any case. Still it should be consistent, producing the same amount of walls if the width is the same.

We could be looking at a variation of [this issue](https://github.com/Ultimaker/Cura/discussions/8965#discussioncomment-250531) or perhaps the problem is that the [hole is misaligned by Arachne](https://github.com/Ultimaker/Cura/discussions/8965#discussioncomment-267352).

Ghostkeeper, 2021-02-24:
Fixed here: https://github.com/Ultimaker/CuraEngine/pull/1415

closed 2021-02-24 by Ghostkeeper

rburema, 2021-02-26:
@Ghostkeeper  I don't think that is the same issue. That offset occurred consistently, not in the middle of a wall. You're thinking of this fix: https://github.com/Ultimaker/CuraEngine/pull/1405

Still closed though, of course :-)

Ghostkeeper, 2021-03-04:
No, I was thinking of Ultimaker/CuraEngine#1415 since the issue here was really the result of that hexagon-shaped hole not being aligned properly in the middle, causing one side to be printed with a single line but the other with two lines.
Combined with the Outer Wall Wipe Distance problem, this was resulting in the blobs the user probably saw in their print.