Ultimaker/Cura #9078: Bitmap import: translucency and linear mode

JamesSF69, 2021-01-06:
**Application version**
4.8

**Platform**
Win 10 nVidia RTX

**Printer**
Doesn't matter

**Reproduction steps**
Importing bitmaps for Lithosphanes

**Screenshot(s)**
[image]

**Actual results**
Linear or translucency are the same results.

**Expected results**
Not too sure, there is no documentation. But I'd assume some difference. 

**Project file**
This was the test file I used.
[image]

**Log file**

**Additional information**
Just wondering if this is supposed to be this way or just why is there the two options? Is one better at a certain "color" image?

fieldOfView, 2021-01-06:
In your screenshot, though hard to really judge, the two don't look exactly the same. The dark grey (second column in the image) looks thicker in the right than in the left. AFAIK, the option you are wondering about makes Cura interpret grey values differently. The Linear option is the most like you would expect it (ie: 50% grey is 50% of the height of the black), whereas with the Translucency option optimises thickness for actual translucency (ie: 50% grey is 50% more translucent than the black column). The theory is that the amount of light that comes through the model does not change linearly with the thickness of the model.

JamesSF69, 2021-01-06:
Good to know. I kinda get it.... Is there a good reason why one would use one over the other? 
Also when these two objects are placed on top of each other they pretty much line up. But yes I guess there is a slight variation. but at this minute difference doesn't it make any difference?
Sorry to go on about this, as mentioned I can't find any real documentation online about this. Thanks, by-the-way!

fieldOfView, 2021-01-06:
In theory, the "Translucency" option produces a better lithophane effect when held against the light.

Ghostkeeper, 2021-01-18:
The difference is hard to see at that scale in Cura, but quite easy to see in the actual print when held against a backlight. The difference is easier to see in Cura at a greater Z scale. The scale is such that it's useless for a lithophane then, but in an actual lithophane the difference is quite visible (your eyes are quite sensitive to the difference).

Here I've scaled the object to 40mm height. The blue object in the back is Translucency. The yellow in the front is Linear.
[image]

The Translucency scale is logarithmic according to a physical model. This is designed to look better when held against the light. The Linear scale is precisely interpolating between the base thickness and the total thickness, and would be more useful for e.g. a height map.

closed 2021-01-18 by Ghostkeeper