Ultimaker/Cura #9079: Print profile is "Not supported", cant create custom profiles or edit new ones

hitech95, 2021-01-06:
**Application version**
4.8.0

**Platform**
Windows 10 2004
i7-4790k, GTX1060

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Install Cura
2. Select the printer in the Setup wizard
3. Create a ner Material or use "PLA+" from eSUN.
4. Try to create a print profile.

**Screenshot(s)**
[image]
[image]

**Actual results**
Print Profile settings are grayed out, cannot store a custom profile.
Default profiles cannot be selected.

**Expected results**
Beeing able to create a custom priofile is the default one are not available.

**Project file**
Not needed, this happend even without any file open

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5776147/cura.log)

**Additional information**
Other people are facing the same issue.
https://community.ultimaker.com/topic/33336-non-generic-material-selection-causes-profiles-to-show-not-supported/

MarkMan0, 2021-01-07:
Facing the same problem, I've created a printer def file, extruder def file and simple quality file, but when I add the printer in cura 4.8, none of the materials are supported.
Here are the files:

**myprinter.def.json:**
```
{
    "version": 2,
    "name": "myprinter",
    "inherits": "fdmprinter",
    "metadata":
    {
        "visible": true,
        "author": "me",
        "manufacturer": "me",
        "file_formats": "text/x-gcode",
        "preferred_material": "generic_pla",
        "has_machine_quality": true,
        "has_materials": true,
        "has_variants": false,
        "supports_usb_connection": false,
        "machine_extruder_trains":
        {
            "0": "my_extruder_0"
        }
    }
}
```
**my_extruder_0.def.json:**
```
{
    "version": 2,
    "name": "Extruder 1",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "myprinter",
        "position": "0"
    },

    "overrides": {
        "extruder_nr": { "default_value": 0 },
        "machine_nozzle_size": { "default_value": 0.4 },
        "material_diameter": { "default_value": 1.75 }
    }
}
```
**myquality_cfg.cfg:**
```
[general]
version = 4
name = Normal
definition = myprinter

[metadata]
setting_version = 16
type = quality
quality_type = normal
weight = 0
global_quality = True

[values]
layer_height = 0.14
```

**Expected outcome:** I believe these to be correct, and this printer should have 1 default profile, which supports all materials.

Of course this is a simplified version of the files, but nevertheless, these should work too.

fvrmr, 2021-01-07:
Did you change some print settings (like layer height) before you tried to create a profile?
Because after this you can create a profile even though it is not supported.

MarkMan0, 2021-01-08:
> Did you change some print settings (like layer height) before you tried to create a profile?
Because after this you can create a profile even though it is not supported.

Not sure I understand this.
I can create custom profiles for the printer and I can slice, but the materials in the material selection are all orange. 
I tried to create this *myprofile*, I tried to do it based on the **alya3dp.def.json** file. That has custom profiles and can select multiple materials. 
I don't know why my custom profile can't do that. 

Is there a property that I overlooked which would enable the support of all materials, similar to `exclude_materials`? 
Or is there anything wrong with my profile?

Ghostkeeper, 2021-01-08:
> "has_machine_quality": true,

This would be why there are no profiles available. You claim your printer has printer-specific profiles but then don't provide any profiles. It'll list any quality profile that has `machine=myprinter` in its metadata, but you don't have any. Your issue is different from what the original poster's issue was. Completely different reproduction steps.

Fvrmr's response applied to the original issue.

Zer025, 2021-12-18:
I have the same problem: when creating a new material, if Material Type is PLA+ (or Nylon, PC, PVA and many others), Cura gives error "Not Supported". To workaround this problem I just used PLA instead of PLA+
I have Cura 4.12.0

Ghostkeeper, 2021-12-29:
For clarity, the problem reported here is NOT that the print profile is "Not Supported". That part is correct and expected. The problem is that the user is not able to then create custom profiles.

We haven't been able to reproduce that issue so far. I hope we can get some clarity on how to reproduce that.

hitech95, 2021-12-29:
> For clarity, the problem reported here is NOT that the print profile is "Not Supported". That part is correct and expected. The problem is that the user is not able to then create custom profiles. [...]

Unfortunalty I'm not able to replicate the issue. I must have re installed cura in meanwhile.

What is that ""Not Supported" means!?

Ghostkeeper, 2021-12-29:
Basically, it means that Cura has no profiles for the combination(s) printer+nozzle+material-type you're using. Instead it will use the default for that combination.

A related issue is https://github.com/Ultimaker/Cura/issues/8229, which started out similar to this one but is currently about the unclarity of making profiles depend on the material type.

closed 2024-12-08 by github-actions[bot]