Ultimaker/Cura #9080: Prime tower does not prime

rlcrom, 2021-01-06:
**Application version**
(The version of the application this issue occurs with.)
4.8
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
windows 10
**Printer**
(Which printer was selected in Cura?)
Tenlog hands 2
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
Select to use a prime tower. I see the shadow, but it doesn't print the tower
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**

(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-07:
I need more information to look into this. Can you share your project file?

rlcrom, 2021-01-07:
Here you go. I think I have the tower working, however, the tower only
printed the two colors once the both colors were needed. So it printed the
main color and built the tower but had no second color until the tower was
a 1/2" high. Then hte second color started to print and had no base to
print to. I can send a picture of that if you need it

Ron

rlcrom, 2021-01-07:
Picture of tower

Sent from a national champion's iPhone 🎱

fvrmr, 2021-01-08:
Can you add your file and picture in Github?

rlcrom, 2021-01-08:
How do I do that? I have another project that is not printing the prime tower this morning

fvrmr, 2021-01-08:
You can add a .zip with your .3mf file in a comment.

rlcrom, 2021-01-08:
[ForSupport.zip](https://github.com/Ultimaker/Cura/files/5787111/ForSupport.zip)

rlcrom, 2021-01-08:
It looks like if I uncheck the alternate mesh removal, I lose the tower and the ooze shield

fvrmr, 2021-01-08:
I see you use an Hands 2 printer, can you send the link where you got the definition for this printer.

rlcrom, 2021-01-08:
I downloaded it from here:
https://drive.google.com/drive/folders/115S2xjY4eGerNBMd40gIr5Vu2XqoKjH7
From Tenlog
My issues seems to be with selecting alternate mesh removal. All the dual extruder stuff like prime tower and ooze shield do not print. I don't know if I'm missing a setting or what but my first prints had no issue

rlcrom, 2021-01-08:
I have the same issue with Cura 4.4 too. Is there a setting that I'm missing?

Ghostkeeper, 2021-01-11:
Tenlog doesn't support Linux, so I can't install that. or @fvrmr it's also not safe to install unsigned executables.

Could you share the printer profiles itself? It will be in your Program Files (where you installed Cura). The files we need are `resources/definitions/Hands2.def.json` and `resources/extruders/tl_dual_0.def.json` and `resources/extruders/tl_dual_1.def.json`.

rlcrom, 2021-01-11:
[profiles.zip](https://github.com/Ultimaker/Cura/files/5795537/profiles.zip)
Here you go

rlcrom, 2021-01-13:
Any updated on this? I closed cura and reopened it and for at least the one model, I could slice with the alternate mesh removal off and I get the ooze shield and the prime tower. Is is days of messing around. Something is not right here. I want that feature turned off to give me better solid color on my model

fvrmr, 2021-01-14:
Still can't reproduce this, need another definitions file. Can you share `tenlog_printer.json` file also.

rlcrom, 2021-01-14:
[supportFIles.zip](https://github.com/Ultimaker/Cura/files/5814032/supportFIles.zip)
I think this is what you want. I'm not sure yet, but I have two printers set up in Cura. I have an ender 3 pro and the hands 2. I have deleted the ender 3 pro and at least the last two slices have worked correctly. Maybe try adding that printer.

rlcrom, 2021-01-14:
I downloaded cura from the Tenlog site. When installed, I ended up with two Cura 4.8 installs. How can I get the hands 2 printer to show up with the normal download of cura?

fvrmr, 2021-01-14:
I finally could reproduce it. And this what I figured out, when you print with two extruders you have can have an ooze shield and a prime tower. These print only when both extruders are active. 
On the screenshot below the prime tower and ooze shield stops because the second extruder is not used for the rest of the print. It only print the M, the rest of the text is blend in the object and is printed by extruder 1. 

[image]

When I move the text a little, the prime tower and ooze shield will print more. Because both extruders are active. 

[image]

So the prime tower and ooze shield stops after extruder 2 has nothing to print. 

I hope this makes sense, if not let me know!

rlcrom, 2021-01-14:
I lose the prime tower and ooze shield if I uncheck the alternate mesh
removal. Do you get that?

rlcrom, 2021-01-14:
Also I don't see the two different color/extruders if I uncheck that box. This is my main issue

fvrmr, 2021-01-14:
> I lose the prime tower and ooze shield if I uncheck the alternate mesh removal. Do you get that?

I have alternate mesh is disabled. And I see the prime tower and ooze shield...
[image]

Does this happen with our Cura version or the other one downloaded from Tenlog?

rlcrom, 2021-01-14:
I think I get this in Tenlog version. I only use that because it has the tenlog printer for me. I don't see that tenlog if I install the ultimaker version

rlcrom, 2021-01-14:
How can I get the tenlog printers in the Ultimaker version?

fvrmr, 2021-01-14:
> I think I get this in Tenlog version. I only use that because it has the tenlog printer for me. I don't see that tenlog if I install the ultimaker version

Ah okay. Then I can't help you with this and you'll need to go to Tenlog for this issue.

But I got it working. So you can download our version of Cura 4.8 and place the printer definitions: `Hands2.def.json` and `tenlog_printer.def.json` in `C:\Users\<your username>\AppData\Roaming\cura\<cura version>\definitions`
And your extruder definitions: `tl_dual_0.def.json` and `tl_dual_1.def.json` in `C:\Users\<your username>\AppData\Roaming\cura\<cura version>\extruders`

rlcrom, 2021-01-14:
Thanks, anything special about uninstalling cura for the new version?

rlcrom, 2021-01-14:
Okay, I installed your version and brought in the files that you ask me to and it worked for my first print. I added the ender 3 to print something there and switched back to the hands 2 and now it doesn't give me the ooze shield or the print tower. My model looks all blue. The color of the first extruder

rlcrom, 2021-01-14:
To add to this. I deleted the ender 3 pro printer and closed/reopened Cura and it works again.

fvrmr, 2021-01-15:
> Okay, I installed your version and brought in the files that you ask me to and it worked for my first print. I added the ender 3 to print something there and switched back to the hands 2 and now it [...]

Yeah, if you switch between printers, especially between printers between with one extruder (Ender 3) and two extruders (Hands 2). You need to click on the object you want to print with extruder 2. When you slice it, you will see the prime tower and ooze shield.

rlcrom, 2021-01-15:
All I did was swapped out one stl and sliced and I have no tower or ooze shield
[image]

rlcrom, 2021-01-15:
I really need this to work correctly. One time it works, another it doesn't. At least I wish I knew what to do to get it to work

rlcrom, 2021-01-15:
How do I get the "blue" to not cross through the green?
[image]

fvrmr, 2021-01-15:
Is this the same project file as before? If not can you share this project file to?

rlcrom, 2021-01-15:
[krypo.zip](https://github.com/Ultimaker/Cura/files/5821648/krypo.zip)
 Here you go. I might add, that I import two stl files. One that is the base model and the other is the lettering. I then slide the lettering into the base model and slice.

rlcrom, 2021-01-15:
Thanks, but then I'm back to having alternate blue and green layers in the green section. I'd like to have the green area as solid as possible to give good color. If I alternate the layers, I get a light green color due to it like striping. If I select the remove mesh intersection and uncheck the alternate mesh removal, I lose the green completely along with the ooze shield and prime tower . try that on your side. This is my original problem

[image]

Ghostkeeper, 2021-01-18:
Basically what you have is two models where one completely overlaps the other (or at least, almost completely such that the remainder is too thin to print). Your problem is not that it doesn't create a prime tower. It's that it completely skips one of your models since it's being suppressed by the overlap with another model.

Cura doesn't know that you expect the model with the letters to override the bigger model. There are three options that Cura could do:
* Simply print both. That's what happens if you disable Remove Mesh Intersection. You get overextrusion and badness overall. Don't do this.
* Alternate which of them gets printed from layer to layer. That's what happens if Alternate Mesh Removal is enabled (which is default). This makes the two models interweave and creates a very strong connection. It's not what you want though since it makes the colour appear striped. Strong connection isn't your problem.
* Print one of them over the other. This is the solution that you're looking for to make the letters come out in a different colour. Cura chooses one model arbitrarily to print over the other (I think the one you load in first?) and in your case it chose the wrong one. You can customise this by giving your priority model a lower "Mesh Processing Rank". This is a per-object setting. This is, I think, the solution you're looking for.

So you'd go into the per-model settings tool on the left, click on "Select Settings". Search for "rank" and enable the one you see then. Then give it a small negative value, to be lower than the rank of the other model (which defaults to 0).

[image]