Ultimaker/Cura #9081: Advanced wipe, coast, combing

Julius971, 2021-01-06:
**Is your feature request related to a problem? Please describe.**
Stringing, caused by melting printed filament during retract

**Describe the solution you'd like**
The idea is that the retraction move is during a wipe move or similar or even the travel move itself.

**Affected users and/or printers**
Fdm printers

fvrmr, 2021-01-07:
What do you mean with a wipe move?

We have something familiar and that is post processing scripts - retract continue. 
You can find this in: Extensions - Post processing - Modify G-code - Add a script - Retract Continue.

Julius971, 2021-01-09:
The problem is, that the printhead stops during retract moves. And i wonder if it could retract while moving.

Julius Maetzig

Ghostkeeper, 2021-01-18:
The Retract Continue post-processing script does this. Just set the normal retraction length to 0mm though.

closed 2021-01-18 by Ghostkeeper