Ultimaker/Cura #9083: [4.8.0] Cutting Mesh produces thin walls

intercipere, 2021-01-07:
**Application version**
4.8.0

**Reproduction steps**
1. Load model
2. Place Modifier Mesh on model
3. Select 'Modify Settings for Overlaps' and 'Cutting Mesh' for the Mesh Modifier

**Screenshot(s)**
[image]

**Actual results**
Cura will partially produce thin walls on the modified mesh, and also change Acceleration settings as seen in the image, even if no parameters have been selected or changed for the cut mesh. It will also print exactly as Cura shows it, it is not a display bug. Changing Line Width, Wall Thickness, Flow etc has no influence on the result.

**Expected results**
The wall thickness and Acceleration should not change if no parameters have been changed for that.

fvrmr, 2021-01-07:
Can you share your project file? Because I can't reproduce this right now.

intercipere, 2021-01-07:
Oh sorry, here it goes:

[test.zip](https://github.com/Ultimaker/Cura/files/5783105/test.zip)

fvrmr, 2021-01-08:
Thanks, I got the same issue. Will discuss this with the team.

But found a workaround, search in settings for bridge and disable bridge settings. 

[image]

intercipere, 2021-01-08:
Thank you, the workaround works for me aswell.

fvrmr, 2021-01-11:
We discussed it with the team. Bridge is in experimental mode, it is not a big issue for now. Luckily you have a workaround for future projects.

Triangulix, 2022-09-27:
In cura 5.1.0 this Bug ist still present.

closed 2024-12-08 by github-actions[bot]