Ultimaker/Cura #9084: Print simultaneously for dual independent head printers

xz6lkn, 2021-01-07:
Printer: Tenlog TL-D3 Pro (dual independent head printer)

Problem: I would like to print two different parts simultaneously, each part printed by one of the two independent heads

Proposed Solution: New Special Mode that allows both heads to print simultaneously (DIHP Simultaneous Mode)

Cura already allows you to select which printer head prints which part (i.e. head 1 prints part 1 and head 2 prints part 2) AND Cura also checks for head clearances to ensure no head collision during printing.

So the building blocks are there, just need a special mode toggle that enables this.

Realistically, at the end of each print layer, one of the two heads will have to wait for the other to finish before both being able to move on to the next layer (because they could be printing completely different parts). I'm OK with this small wait time because the time savings is still there by being able to print two different parts at once.

BTW, the Tenlog printer already has a "Duplicate" mode which allows simultaneous printing using both heads at once, but you of course can only duplicate a single part using this mode.

***This request is not satisfied by the current "Print Sequence" setting because it requires one of the two heads to be disabled to use this feature.

fvrmr, 2021-01-07:
We can't support this, because Ultimaker printers don't have two independent printer heads.

xz6lkn, 2021-01-07:
Sounds like a good solution -- I've been coding since 1979, so no problem there.

closed 2023-07-13 by github-actions[bot]