Ultimaker/Cura #9086: Cura 4.8 inverted axis

shawnp30, 2021-01-07:
I have an issue. When you using cura 4.8 my prints are inverted when printing. In cura everything looks as is should. When modeling a calibration cube it shows x and y in the correct axis. When printing x is in the back of the model I am using marlin. Corexy blvcube printer with gtr v1.0 mother board. I have switched y endstop from min to max and changed it in firmware. I’m still getting inverted/mirrored prints. What am I doing wrong.

GregValiant, 2021-01-08:
If the print head moves along the beam it's mounted on, what axis is that?

shawnp30, 2021-01-08:
X axis

GregValiant, 2021-01-08:
Ok.  So if you were standing on the left side of the printer (as shown in the video) the 0,0,0 is in the left front corner.  That is as it should be.  A Cura Gcode file (sliced with the printer setting "Center Origin" turned off) will contain all positive X and Y values.  The easiest way to mirror, is to multiply each of the numbers on one axis as it arrives in the printer by -1 and move (virtually) the origin of that axis to a different corner so the negative numbers stay on the print surface.  Does it look like that is what's happening?  I think it would need a menu option to do something like that (some IDEX printers have the option).  When you tell the printer to go to X10, Y10 is it still near the origin corner?  What I'm trying to figure out is how the printer could mirror something just by itself.  The numbers just don't work out for me.  Are you sure it isn't a rotation?  If you were to print a small character with it's right arm in the air would the print end up with the left arm in the air(?) or would it just be facing a different direction.

Ghostkeeper, 2021-01-18:
I've seen sometimes that a printer apparently had a negative steps per mm. Note that Cura doesn't use this setting at all since it doesn't care about step size, unless you use the X3GWriter. But if the firmware has a negative steps per mm in one axis, the print will get mirrored by the firmware across that axis.