Ultimaker/Cura #9087: Unneeded support is created, even if blocked out on some files

SofaKingRotten, 2021-01-07:
**Application version**
Started in version 4.2.1 to current 4.8.0

**Platform**
Windows 10 Pro x64, 16Gb RAM, Intel Core i5-3570 3.5Ghz CPU

**Printer**
Wanhao Duplicator 6

**Reproduction steps**
1. Add support in print settings, Grid and Lines will surely fail, others untested.
2. Slice and preview. The support is included where it is not needed.
3. If you add blocking, unneeded support is still provided unless you block all support-necessary areas of model.

**Screenshot(s)**
[image]
[image]

**Actual results**
On a straight, cylindrical, upright column without any overhang, unnecessary support is created on some models or files.

**Expected results**
The support should only be provided as in picture 2.

**Project file**

[unneeded support.zip](https://github.com/Ultimaker/Cura/files/5783651/unneeded.support.zip)

**Log file**
No crash or similar issue occurred.

**Additional information**
None.

StuSerious, 2021-01-08:
Hey there!
I tried to replicate your results following your steps and settings, but everything looks fine on my end!
[image]
I suspect that it could be an error on you side, what build of Cura are you using?
Also, could you please provide your project file? [File > Save Project]
Thanks!

SofaKingRotten, 2021-01-08:
I am using Cura 4.8 but I checked all versions back to 4.0 and 4.0 didn't exhibit this problem. I did leave the project file of that part, but here is another part and now this is happening on version 4.0 too. Here is pic and project file of this new part, exported from version 4.0 (this part also fails on 4.8 for me). This part has a small alignment dimple embossed "into" part just above the support that is being created, but is unneeded. There is no overhang of any sort above that support.

[image]

[Selector.zip](https://github.com/Ultimaker/Cura/files/5789346/Selector.zip)

SofaKingRotten, 2021-01-08:
I just re-worked that last part in Inventor and now the support issue is not showing up. It must be a problem with my model part before it gets to Cura. Unless you find something, this case is closed.

StuSerious, 2021-01-09:
Hey there again!
What I meant was a Cura Project File export so that I can also import your settings and printer config to see if the problem is there.
Having a quick look in MM reveals some strange triangle mesh point placements and periodical horizontal lines, nothing too unusual but it's possible the engine got confused.
[image]
Also, this is what MM Inspector spits out 🤣
I have no idea why lol
[image]
Anyways, glad you figured out a solution!

SofaKingRotten, 2021-01-11:
Hi, in both cases I did File --> Save Project and zipped them as image file, not sure how else to do that. I have since looked at my support options in settings and see something called "Use Towers" which is checked by default in 4.8. When I uncheck this option my 'unneeded'
 support goes away. Must be some slight overhang there but the model doesn't show it. Thank you for all your efforts and for pointing me in the right direction!

closed 2021-01-11 by SofaKingRotten