Ultimaker/Cura #9088: The problem of ooze shield for dual extruder printer

ANYCUBIC-3D, 2021-01-08:
Problem:
We have a dual extruder printer. The extruder prints ooze shield first when it starts to print the next layer. The ooze shield cannot wipe the second nozzle when the second is at the same layer as the first nozzle, because the extruder will not switch to the other one when it starts to print the next layer. It affects the printing quality.

The solution we'd like:
We hope that, after the first extruder switch to the second, the second extruder prints the ooze shield first. If the extruder don't switch to the other one at the layer, the extruder prints ooze shield after it prints the current layer.

Ghostkeeper, 2021-01-13:
Hi! Thanks for continuing to maintain your printer definitions :)

> The ooze shield cannot wipe the second nozzle when the second is at the same layer as the first nozzle, because the extruder will not switch to the other one when it starts to print the next layer.

This is a bit of a design flaw that we've never really thought about. The ooze shield was designed to be something you would always hit when switching extruders, because on Ultimaker printers the extruder switch happens in the switching bay on the side of the printer. So for every switch it needs to move back and forth, across the ooze shield. The ooze shield is printed as one of the first parts of a layer, so it's always as high as it can be, and the nozzle always hits the ooze shield at the correct height of the current layer.

The ooze shield is designed to be something that the ooze is wiped off on by travelling across the shield, not as something that the ooze is wiped off on by printing the shield itself.

So if I understand correctly, your problem is that your printer doesn't move across the ooze shield after every switch (`machine_extruder_start_pos_abs` is `false`, which is indeed quite common with many printers) and so doesn't wipe the ooze off then.

Your solution is to print the ooze shield later than what it's currently doing if there is an extruder switch on that layer. It would print the ooze shield on the (first?) extruder switch of the layer, when printers with 2 extruders are most prone to ooze. This is a disadvantage for printers that do have a switching bay though.

If I understood your requirements well though, there are two workarounds I can think of, one of which is an actual fix as far as I'm concerned:
* You could set the extruder start/end positions to be absolute rather than relative, and somewhere on the side of the printer. This will cause the nozzle to move across the ooze shield with every switch.
* You could print a prime tower instead. The prime tower is actually intended for the way in which you're using it as one of its functions: to purge excess materials. The prime tower even has an extra function to wipe off the inactive nozzle on the prime tower as well if not all your nozzles have the same position.

Let me know if we understood your problem correctly. It's quite complex and it's hard to visualise from your short story.

imagesurgery, 2023-08-28:
I'm a new user of a Snapmaker Artisan, which has a DEX that functions similarly to the OP. I would like to thank you for the clearest description of why my ooze shield is failing to provide any benefit, and why my extruder is switching in the middle of my prints. Not good, lots of blobbing, un-primed nozzles, etc.

I will try purge/wipe towers as it sounds like that may solve the problem, for the most part, but I'd like to request that this feature be improved as the number of DEX printers of my type is increasing, and others may be caught out like I have been. Although my knowledge is limited and I'm certainly not the expert here, I'd like to propose a method I'd really like, and hopefully you'll be able to implement in a much improved manner (over my suggestion, that is!)

Consider re-working the ooze shield feature to perform the nozzle switch over the ooze shield itself, so that priming and printing first happens on a layer at the ooze shield, then moving into the print itself. There's some solid benefits of ooze shields over prime towers, first and foremost being that the prime tower must be described manually, including any calculations about what it needs to do, where it is, etc.; ooze shields perform this similar function automatically, removing the space for user error, old settings from the last print, and other issues which the manual nature of prime towers doesn't address.