Ultimaker/Cura #9092: larger than recommended layer heights cause the cura engine to crash

ruckusmattster, 2021-01-09:
**Application version**
arachne cura

**Platform**
windows 10-integrated intel graphics on mobile i3

**Printer**
CR10s Pro

**Reproduction steps**
1. increase layer width to 1.6
2. layer height to 1.28 
3. enable spiralize outer contour
4. enable smooth spiralized outer contours

**Screenshot(s)**
just a error message showing cura engine has stopped respondig

**Actual results**
cura engine must be closed and then you can resume use as usual (as long as a new layer height is specified)

**Expected results**
the model should slice as normal (works in cura 4.8)

**Project file**

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-15:
We have a lot of issues for Arachne, see the discussions here: https://github.com/Ultimaker/Cura/discussions/8965 
We hope to solve a lot of issues soon.

Ghostkeeper, 2021-01-18:
I'm unable to reproduce this issue. Here is a screenshot of a model I sliced with your adjustments (with "layer width" I assume you meant "Line Width"):

[image]

I tried 2mm layers as well. All seems to slice fine.

Do you have a project file that reproduces your issue?

ruckusmattster, 2021-01-21:
[Filla_Fella.zip](https://github.com/Ultimaker/Cura/files/5851200/Filla_Fella.zip)
i think this it? i pressed save project and voila

Ghostkeeper, 2021-02-01:
Indeed I could reproduce this on the Arachne Alpha release from December. But using the current state of the branch it no longer crashes. So this has already been fixed since then! :grinning:
It's been more than a month since that alpha release. We've mostly been working on the main 4.9 release (which will not include Arachne yet) but there have been some changes.

Thanks for the bug report nonetheless. It's useful to know that these bugs also occur out in the field.

closed 2021-02-01 by Ghostkeeper