Ultimaker/Cura #9093: Arachne - movement for walls around holes

GregValiant, 2021-01-09:
**Application version**
Arachne 99.9.9.0 downloaded Jan 9

**Platform**
Win 10 64 bit

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Slice a model with vertical holes in it
2. Create Gcode file

**Screenshot(s)**
N/A

**Actual results**
The there are 10 holes in the part.  The print head prints a single loop around each hole and then moves to the next hole and prints a single loop, moves on, and so forth.  It makes 3 circuits around the entire part to complete the 3 walls around the holes.  There would seem to be a lot of unnecessary movement as all layers show the same single loop-and-move pattern.

**Expected results**
Like in previous versions it should print all the walls around each hole in turn to minimize travel.

**Project file**
3mf file and a gcode file.
[HoleBrkt.zip](https://github.com/Ultimaker/Cura/files/5791271/HoleBrkt.zip)

**Log file**
N/A

**Additional information**
I installed the Arachne version I downloaded today.  My previous installation (10 days ago?) had the same movement pattern between holes.

rburema, 2021-01-12:
Hi @GregValiant  Can you try disabling 'Order Inner Walls By Inset'? That should stop Cura from acting the way you describe. If it was for models like this only, that (disabled) should probably be the default even, but on models with holes closer together, so the walls 'bleed over' into each other, it may not be what you want.

Both that and to a lesser extent 'Outer Before Inner Walls' have different behaviours in Arachne than in their 'plain' counterparts. We should have mentioned this in the change log.

GregValiant, 2021-01-13:
I didn't see that one.  Mentioning it in the change log would be nice.  I wouldn't have read it, but it would be nice.  Thanks.
I believe this can be closed.

GregValiant, 2021-01-13:
Settings.

closed 2021-01-13 by GregValiant

Ghostkeeper, 2021-01-18:
It was mentioned only with the words "and behaviour changed" in the change log. It should get a more prominent spot in the change log for the beta!