Ultimaker/Cura #9094: Cannot enable `Support Structure` setting (cannot change support structure type)

stevebroskey, 2021-01-09:
**Application version**
4.8.0

**Platform**
Ubuntu 18.04.5 LTS

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Open model
2. Click Gear beside `Support` in `Settings`
3. (Preferences opens with Setting Visibility)
4. Enable "Generate Support"
5. Try to enable "Support Structure" (`info` Icon never disappears, tooltip says `This setting has been hidden by the value of Generate Support. Change the value of that setting to make this setting visible.`)

**Screenshot(s)**
[image]

**Actual results**
No change, `Support Structure` setting not visible.

**Expected results**
`Support Structure` setting should be visible.

**Project file**
N/A

**Log file**
```
2021-01-08 20:42:18,788 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-08 20:43:16,786 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:31:42,790 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:31:52,793 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:32:13,789 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:32:49,788 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:38:05,787 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:39:09,789 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:45:46,791 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 08:46:04,789 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
2021-01-09 10:50:18,789 - DEBUG - [MainThread] cura.AutoSave._onTimeout [59]: Autosaving preferences, instances and profiles
```

**Additional information**
(Extra information relevant to the issue.)

stevebroskey, 2021-01-09:
I also get these errors on startup (using `cura &`):
``` 
$
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
```

Not sure if this is relevant or not, but the `qml: TableViewSelection` could be relevant to settings tables not showing properly(?).

fieldOfView, 2021-01-09:
Did you enable supports for your print? Unless supports are enabled, changing the settings that control how supports are created makes little sense.

stevebroskey, 2021-01-09:
> Did you enable supports for your print? Unless supports are enabled, changing the settings that control how supports are created makes little sense.

Yes, supports were enabled. I toggled that as well as several other options during troubleshooting.
Embarrassingly, after checking the logs and config files by hand, the settings showed after a restart of cura. 

Closing this issue.

closed 2021-01-09 by stevebroskey