Ultimaker/Cura #9095: [Enhancement request

Sophist-UK, 2021-01-09:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

fvrmr, 2021-01-15:
We will discus this with the team. Keep you posted.

fvrmr, 2021-01-18:
We will defer this for now, because we think this is not one of our priorities right now. But we are experimenting with some alternatives to improve the transition in flow rate.

closed 2023-07-13 by github-actions[bot]