Ultimaker/Cura #9096: Multiply Models does not work with Support Blockers

KenLucke, 2021-01-09:
I encountered this when trying to create a model with samples of all the Infill types, which required 13 "support blockers" (worst name ever, as they are far more versatile than that) of identical size, shape, and top layers (0), with only the infill pattern different.  No matter what I did - trying to duplicate the brand new SB, trying to duplicate it after moving/resizing it, or trying to duplicate it after choosing which settings were to be adjusted, nothing allowed me to duplicate/multiply it.  

I recall that this was also an issue a couple of times before when I was trying to create a pattern of actual support blockers on a model.

There are often times one might want identical support blockers in a model/project, not being able to simply duplicate them hinders the efficient setup of a model for slicing.

**Application version**
4.8

**Platform**
Mac OSX 10.15.7

**Printer**
n/a

**Reproduction steps**

1. Load a model
2. Create a Support Blocker, with or without using it as a settings modifier
3. Select the Support Blocker, either by clicking on it, or clicking on its name in the object list
4. Try to duplicate the Support Blocker using "Multiply Model"

**Actual results**
Nothing

**Expected results**
Support blocker should be duplicated [multiply model] times.

The model after MANUALLY placing, sizing, and setting all the Support Blockers (doesn't show the issue, only what I was trying to accomplish)
[image]

konskarm, 2021-01-13:
If I understand correctly, you used the Support Blocker (Eraser) as a Cutting mesh. This is allowed, but it is not the normal workflow for what you want to achieve. 

In general, support blockers are bound to the object you create them on. Hence why you cannot really duplicate the support blocker itself without duplicating the object it is bound on (it has no parent then). Try adding a model and a support blocker on it, and then duplicate the model. In this case, both the model and ALL of its support blockers will be duplicated with it.

If we decide to allow Support Blockers to be duplicated, then they will just end up being bound to the same parent. This means that if you delete or move the parent, the support blocker will follow, which can lead to a lot of confusion. Imagine the following scenario:

- You add Cube A and Cube B
- You add a Support Blocker in Cube A (let's call it Eraser A1), which you convert to a cutting mesh so that it modifies Cube A
- You multiply the support blocker, creating an Eraser A2 with the same modifiers
- You move Eraser A2 on Cube B to modify it
- You remove Cube A 

**Result**: Cube B is no longer modified by the cutting mesh "Eraser A2".

Normally, for your situation you should import a cube and convert it into a cutting mesh. If you do that, the cube won't be bound to any other object and therefore it can be freely duplicated, also copying all the settings that you change for it along the way.

That being said, I will discuss the possibility of duplicating a support blocker with the team.

KenLucke, 2021-01-13:
OK, understood, but in this case (and other cases to which I was referring), the support blockers _were_ being bound to the same model, just altering different parts of it.

konskarm, 2021-01-14:
We discusses with the team today and it seems that this is a known issue (https://github.com/Ultimaker/Cura/issues/8776), so I'll close this ticket in favor of the other one. Please add any extra comments there.

Also, although not prioritized high, it is already in our backlog (Devs see CURA-7851).

closed 2021-01-14 by konskarm