Ultimaker/Cura #9097: Ironing Insets also on conave regions

koelewij, 2021-01-09:
**Application version**
4.8.0

**Platform**
Win10 X64 

**Printer**
CR-6 SE

**Reproduction steps**
1. Enable Ironing on a part with pockets
2. After slicing the Ironing is inset away from the wall when the bottom of pocket is ironed.

**Screenshot(s)**
[image]

**Actual results**
Ironing is inset away from the wall when the bottom of pocket is ironed.

**Expected results**
Ironing could go at least up to the wall, an possibly a bit underneath. 

**Project file**
N/A

**Log file**
N/A

**Additional information**

fvrmr, 2021-01-12:
You can change the ironing inset to 0.0 mm and than the ironing isn't away from the wall. 
[image]

koelewij, 2021-01-12:
That option works fine for the surfaces with a wall, but in this case also the inset on the surfaces without a wall is set to 0 causing a ragged edge on the edges.
[image]

fvrmr, 2021-01-18:
You can change the inset to the radius of your nozzle. 
See picture below. The red line is the distance you can have from your nozzle till the edge of your print.
[image]

amstan, 2022-01-24:
Regardless of what we set the inset setting to, it should only be applied to stay away from the top of the red lines. I would actually want it to go extra on top of the yellow areas when the next layer is supposed to have more on top, so there is no unironed seam in the concave corner.

[image]

Blue distance should be what this "Ironing Inset" setting controls, purple distance (conave regions) should be 0 or even negative (to go under that red line to the right of it (that one's on the next layer, on top of the yellow region)

koelewij, 2022-01-24:
@amstan 

I totally agree on that. 
Also in this case where the floor surface is twice the nozzle width, i don't get any ironing. 
[image]
So as mentioned above, the purple offset should be 0 or negative.

koelewij, 2022-01-24:
Here a Slicer that does what i expect it to do. 
[image]