Ultimaker/Cura #9098: Feature Request: Vary support extruder with hieght

mine2209, 2021-01-09:
**Is your feature request related to a problem? Please describe.**
I use support interface of PETG when printing PLA to get great surfaces without expensive filament, at some points in the model I want PLA support as I need the model to be attached to the bed still and PLA PETG dont stick.

**Describe the solution you'd like**
support interface extruder to be added to the change with height feature in post processing or similar option or ability to select different extruder for support interface on supported bridges

**Describe alternatives you've considered**
At present I manually open the Gcode and delete temperature changes and swap T0 and T1 commands with find feature of a text editor it works but is annoying as hell to do and time consuming if want a few swaps

**Affected users and/or printers**
Everyone printing dual extruder who wants more control or to be able to choose a petg support to save cost and hassle of PVA supports where they dont need them to dissolve

**Additional context**
(Add any other context or screenshots about the feature request here.)

fvrmr, 2021-01-15:
I'll discus this with the team. Keep you posted.

fvrmr, 2021-01-18:
Can you give us a better explanation, maybe a drawing. So that we can understand and see what you exactly want.

mine2209, 2021-01-18:
Here is a gear example with a few images from the slicer, I print these lathe change gears in PETG, these are large gears and require almost a full support layer low down as only the centre hub is on the bed, and then a large bridge further up the print. The bridge in question is too large for PETG at an acceptable surface finish so needs supporting.

[image]

At the first support interface layer only the centre of the gear is on the bed the rest of the model will only adhere to the bed through the support so I need to use PETG supports as I require the adhesion to prevent warping of the gear on the edges. As the support is only on the outer edge of the gear and as it is a small surface area which I can access from the edge it is relatively ok to remove the support from.

Later in the print at approximately layer 13 in this case I need the support interface extruder to change to E2 (PLA). As I need a large recess in the face of the gear to be supported, if I use PETG here as the surface area is so large and internal (cant get blade in from the side) it is nearly impossible to remove without damage to the surface as the PETG adheres too well. If i use PLA for the support interface here then the face is smooth and the support is very easily pulled off so ideal.
See next image to see this layer

[image]

At present to achieve this I have to set the support interface extruder to E2 (PLA) and then open the gcode file produced by cura and manually remove the temperature changes to cool down E1 and heat up E2 and commands to swap extruders for the first support interface and leave the ones for layer 13 as E2. Obviously this is a pain to do and open to error on my part and very frustrating if needing to tweak settings as I have to repeat the whole process over again.

What I would like is for change at Z to have the option to change the support interface extruder, so I can just use E1 for support interface up to say layer 6 and from then on use E2 for the support interfaces and not to have do the gcode edits.

I am happy if it isnt in the change at Z drop down menu just anyway of swapping which support interface extruder is used where would be perfect.

If this still doesnt make sense let me know and I will produce a video to hopefully show the issue better.

Many thanks again.

Sam

skylineboy200000, 2021-01-18:
I understand this use case I think, essentially the ability to specify which extruder prints support material based on a specified or multiple different layer heights. This gives more control over which supports are using E1 vs' E2 vs a combination of the two?

jellespijker, 2021-01-19:
Hi @mine2209 ,
If I understand correctly you want a PLA interface for your support at the inside (for easy removal) and PETG on the outside (for beter adhesion).

If I draw a cross-section it would look something like this
[image]

I can certainly see the value in something like that. But I'm not sure if you want it in the change-at-Z script. I personally think it would we better to allow a user to define the interfaces for a custom support mesh.

mine2209, 2021-01-19:
What you show there would be a great feature but I would expect it would take alot of development and time to do and refine.

I really would be happy with varying it by layer height as I say at the moment, in terms of development effort vs reward 99% of the cases I have done this for I have achieved great results from changing based on specific Z height range.

It appears changes to the way in which the slicer manages setting is already variable by height, just like changing speed or retraction settings by Z height a drop down box for changing which extruder does support interface and/or support structure is all I want.

Another use case example including amazing artwork... PLA model that I am happy to largely support with tree supports from the buildplate. At the top there is complex geometry that really would benefit from PVA supports. Rather than having to use PVA for all supports on the model, I am requesting to be able to use PLA (E1) support low down and then at set layer height I would be able to change to using PVA (E2), later on in the model I may wish to change back again, saving cost of filament and time printing slower soluble supports where I dont need them

[image]

Using this same example I would be able to use PVA support interface the whole model but keep the support structure as PLA lower down in the model where it can fall away but in the complex geometry higher up where PLA support structure would not be able to fit out of the gaps in the model I could change to using fully dissolvable support structure, saving cost and time.

And all accomplishable through Z height which surely would be easier to implement to start with than complex feature