Ultimaker/Cura #9099: after "select face to align build plate", object does not touch the buildplate

Jogge92, 2021-01-09:
**Application version**
Cura4.8.0

**Platform**
Windows10

**Printer**
UM2+

**Reproduction steps**
1. Import STL FIle
2. Select face to align to build plate

**Actual results**
[image]

**Expected results**
[image]

fvrmr, 2021-01-11:
I can't reproduce this issue, when I try this my Z = 0. Can you share your model or another model that has the same problem?

Jogge92, 2021-01-11:
I cant upload the original file, sorry...but this file hast the same problem!
the surface i want to align is still the as in the pictures. thx
[buggy_part.zip](https://github.com/Ultimaker/Cura/files/5797706/buggy_part.zip)

konskarm, 2021-01-12:
I cannot reproduce that either. I tried with your stl file, but whatever face I select to align it ends up properly at Z=0.

closed 2021-01-26 by no-response[bot]