Ultimaker/Cura #9100: Z seam alignment

Julius971, 2021-01-09:
**Is your feature request related to a problem? Please describe.**
If the seam is located on an overhanging corner it looks terrible.

**Describe the solution you'd like**
Id like the z seam alignment options to get more options like, prefere straight side on overhang, where the seam will be placed on straight walls when the f.eg. Sharpest corner is over 45deg. Overhung.

**Affected users and/or printers**
Fdm

**Additional context**
I saw the negative effect of the seam by printing benchy. When cura placed the seam on a corner on the lower side of the boat it looked like crap.

fvrmr, 2021-01-15:
Did you look at the option `seam corner preference`. Maybe that will help you with the z seam. Let me know if this is a good solution.

karabas2011, 2021-01-16:
I have a frequent problem with Z-seam. ( tried all combinations)  It looks like the coasting is turning on but actually off.
I have small gap in point when head finishes external perimater and travels away. 
Cura 4.6.1  0.15mm layers POM 20mm/sec
Looks like some plastic drags with nozzle insteed of adgering.
I need negative coasting values ? to add some plastic when line is finished?

fvrmr, 2021-01-18:
@karabas2011 can please make a new issue for this and fill in the form: https://github.com/Ultimaker/Cura/issues/new/choose.

Ghostkeeper, 2021-01-18:
Duplicate of https://github.com/Ultimaker/Cura/issues/8127 .

Overhanging Wall Angle is the setting that determines how much overhang is too much overhang for the seam to be placed.

closed 2021-01-18 by Ghostkeeper