Ultimaker/Cura #9103: Naming gcode files while saving

hygroscopiccarpaccio, 2021-01-10:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

I normally save all my gcodes for reuse at a later date. When doing so, I sometimes cant remember if the settings were optimal or not, hence naming the gcodes with some information. I'm not into opening the gcode and analyzing it, especially since it's hosted on the octopi.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

Currently when saving gcode files I manually add this to the filename so I can easily print it again
+profile (ie draft/fine)
+material
+infill 
+plate adhesion
+print time

It would be great to have these and more options to be added to the filename automatically. Perhaps also dates and other people might have more requirements.
It would also be great to be able to set the default directory for opening and saving files. -Remembering the last option is not always enough.

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

Doing it manually, but it's not concise 

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

re-printers

**Additional context**
(Add any other context or screenshots about the feature request here.)

Rather than to have a tickbox such as for "adding machine prefix" to the name, I'd like to see an option where you can compose your own prefix. For instance %p-%m-%i-%p-%P (or similar) in the example above. 

THANKS

fieldOfView, 2021-01-10:
There are 2 plugins that will let you do what you propose:

https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/CustomJobPrefix and https://marketplace.ultimaker.com/app/cura/plugins/lokster/CustomSupports

hygroscopiccarpaccio, 2021-01-10:
Great, thanks

KenLucke, 2021-01-10:
Have to second the GcodeFilenameFormat recommendation.  I use it heavily, and the author is highly responsive to feedback.

[image]

nallath, 2021-01-11:
Since there are plugins that can do this, I'm going to close this.

closed 2021-01-11 by nallath