Ultimaker/Cura #9104: Support Type choice in Per Model Settiings

KenLucke, 2021-01-10:
**Is your feature request related to a problem? Please describe.**
Sometimes, various parts of a model, or different models on the same build plate, are better with one type of supports or  another (normal vs. tree).   Unfortunately, you can only choose an overall type for the whole slice.

**Describe the solution you'd like**
The ability to add "Support Type" to the list of things that Per Model settings can accomplish when using a "support blocker" or other model as a settings modifier.

**Describe alternatives you've considered**
The only option is to either print models separately (if possible), or settle for one type of support for the entire model, sacrificing the ability to use the best suited support for a certain area.

**Affected users and/or printers**
Those who are more detail oriented when designing and placing supports.

fvrmr, 2021-01-15:
Will this discuss this with the team, but I think it is a big change. I'll let you know.

fvrmr, 2021-01-18:
Like I already mentioned it is a big change. So we will defer this, it takes a big amount of time for the developers to build this.

jellespijker, 2021-01-19:
As @fvrmr mentioned, this would indeed require a major rewrite of the code. At the moment we have different priorities, therefore we can't honor your request.

closed 2021-01-19 by jellespijker