Ultimaker/Cura #9106: Setting not visible but effective - Connect Top/Bottom Polygons

Paradroid09, 2021-01-10:
**Application version**
Cura 4.8.0

**Platform**
W10

**Printer**
Artillery Sidewinder X1

**Reproduction steps**
1. Adjust "Top Surface Skin Layers" to 1 (or greater), to select diffrent Skin types 
2. Adjust "Top/Bottom Pattern" to "Lines"
3. Adjust "Bottom Pattern Initial Layer" to "Concentric"

**Screenshot(s)**
[image]

**Actual results**
Selecting the "Top/Bottom Pattern" to "Lines" hides the Setting for "Connect Top/Bottom Polygons" but the initial pattern still uses this setting in the background.

**Expected results**
Do not hide the setting, when at least one infill pattern uses. Also for the experimental "Top Surface Skin Pattern" setting.

KenLucke, 2021-01-10:
Did you set visibility of settings to "All"?

fvrmr, 2021-01-11:
Thanks for your issue, we will discuss this with the team.

Paradroid09, 2021-01-11:
> Did you set visibility of settings to "All"?

Hey Luke....
Thank you for your Idea.
I use a custom visibility set and this setting is activated.

But the visibly is enabled and disabled dynamically, according to the other setting, I described.

BR

closed 2021-01-12 by Ghostkeeper

Ghostkeeper, 2021-01-18:
This was an oversight. The setting is affecting and should be affecting the print result, and thus it should also be enabled.

I've made it properly remain enabled in the next release for this case: https://github.com/Ultimaker/Cura/commit/263e1ae3a781dcb2e3991250a8b9c9b8d536e97e

commit 263e1ae3 (2021-01-12): Show 'connect top/bottom polygons' also when initial layer bottom pattern is concentric

Fixes CURA-7962.
Fixes #9106.