Ultimaker/Cura #9107: [4.8.0] OpenGL glitch on M1 MacBook with large project size

jamesloper, 2021-01-11:
**Application version**

4.8.0

**Platform**

2020 M1 Macbook Air

**Printer**

Ender 3 V2

**Reproduction steps**

1. Import 14 STL files to print in one go.
2. Select Standard Quality and change to Gyroid infill
3. Click preview

**Screenshot(s)**

[image]

[image]
In this ss there is actually another bug but not as serious (layer 155 is selected while some layer in the middle is shown)

**Actual results**

Multiple vertex explosions and some serious lag

**Expected results**

The visuals should accurately represent the underlying polygon data.

**Project file**

https://drive.google.com/file/d/1reB8H5fc3PTBRx_UNsxYtokS-lnjEzOY/view?usp=sharing

**Log file**
There was no log file, as in...
```
james@M1 ~ % tail "$USER/Library/Application Support/cura/4.8.0/cura.log"
tail: james/Library/Application Support/cura/4.8.0/cura.log: No such file or directory
```

bkarr123, 2021-01-15:
I'm having this same issue as well. I just tried to slice a file and this is what my preview looks like then Cura crashed when I mess with the preview. 

This has happened with 2 different STL files. Both are single STL files with 1 or 2 objects

Running Cura 4.8.0 on a Macbook Air M1 (2020)

[image]

fvrmr, 2021-01-18:
Cura is currently not supported on new Macbooks containing the M1 chip. Apple introduced incompatibilities between the OpenGL libraries used and the latest firmware used by the Macbook. 
You can read more about the [Apple M1 chip on their website](https://www.apple.com/mac/m1/). 

We look into fixing this in an upcoming version.

bkarr123, 2021-01-21:
> Cura is currently not supported on new Macbooks containing the M1 chip. Apple introduced incompatibilities between the OpenGL libraries used and the latest firmware used by the Macbook.
> You can read [...]

Wow, that's unfortunate. Please keep us posted.

bkarr123, 2021-02-26:
Is there an update on Cura being supported with the Apple M1 chip in the new macbook?

nallath, 2021-02-26:
> Is there an update on Cura being supported with the Apple M1 chip in the new macbook?

Not yet.

closed 2024-12-08 by github-actions[bot]