Ultimaker/Cura #9108: Cubic Subdivision overlaps Extra Infill Walls

JohnTasto, 2021-01-11:
4.8.0
Windows 10
Heavily modified BIBO 2

When `Infill Pattern == Cubic Subdivision` and `Extra Infill Wall Count > 0`, the infill lines and the extra infill walls overlap. It can be compensated with `Infill Overlap [Percentage] < 0`, but that causes a gap if `Infill Pattern` is later changed away from `Cubic Subdivision`.

The following screenshots have `Extra Infill Wall Count == 2`.

`Infill Pattern == Cubic Subdivision`
`Infill Overlap Percentage == 10%` (default, excess overlap)
[image]

`Infill Pattern == Cubic Subdivision`
`Infill Overlap Percentage == -190%` (compensated)
[image]

`Infill Pattern == Cubic`
`Infill Overlap Percentage == -190%` (gap)
[image]

Project file:
[hole-small.zip](https://github.com/Ultimaker/Cura/files/5795243/hole-small.zip)
Printer definition:
[bibent.zip](https://github.com/Ultimaker/Cura/files/5795246/bibent.zip)

fvrmr, 2021-01-12:
Thank you, I got the same. Will discuss this issue with the team.

fvrmr, 2021-01-18:
We will defer this for now, it is not our priority because not many people are using this option.

jellespijker, 2021-01-19:
The bug you reported can indeed be reproduced. But it would require relatively a lot of effort to fix it. At the moment we have a few different priorities. One of these is priorities is the new experimental engine Arachne. The way that infill area's are calculated differs compared with the "old" way 4.8 etc. The part of the code where the surfaces are generated in the new experimental Arachne engine are due to be rewritten. We will keep this behaviour in to account at that time.
For now I will close this issue. Thank you for reporting it tho.

Ghostkeeper, 2021-01-28:
Let's keep these things open then. It's not a priority due to the rarity of the case, but it is a real issue and we should know that it exists.

JohnTasto, 2021-01-29:
Issue 1210 is a question about IDEs from 2016. I don't understand the connection - am I missing something?

Ghostkeeper, 2021-02-05:
Ah, sorry. Broken link. That should be Ultimaker/CuraEngine#1210.

Ghostkeeper, 2021-12-14:
This is now fixed, since https://github.com/Ultimaker/CuraEngine/pull/1210 is merged.

closed 2021-12-14 by Ghostkeeper