Ultimaker/Cura #9109: Request: Horizontal expansion for first layers on support

mine2209, 2021-01-11:
**Is your feature request related to a problem? Please describe.**
Slight elephants footing on the print  where on support

**Describe the solution you'd like**
initial layer on top of support horizontal expansion

**Describe alternatives you've considered**
changing the model and manually adding to my desigsn which I may do in the future but issue comes if I dont have the design files but just other peoples models

JohnTasto, 2021-01-11:
In my experience elephant's foot occurs when the heated bed holds the lowest layers above the glass transition temperature. I'm curious what is causing it for you when printing higher up on support?

mine2209, 2021-01-11:
Ive not noticed it before these prints I assumed its the small Z gap allowing the filament to deposit slightly off to the side due to angle of the support. I noticed it and then was trying to work out how to correct and was all i could think. Part has gone out to client already but next time I print one I will examine it more closely to see if I can see anything that may help

fvrmr, 2021-01-18:
You would expect that the opposite would happen. Because of the space between the model and support you would expect that the material will go down instead of horizontal expansion. 
Here is some more information about Elephants footing: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/elephants_foot.md
Maybe you can find your solution there, because warping also happens on the initial layer.