Ultimaker/Cura #9110: Initial layer - print small features last and print brims outside-in within holes

JohnTasto, 2021-01-11:
**Is your feature request related to a problem? Please describe.**
It is difficult to print short lines with PETG. Except when printing on or adjacent to existing PETG, the first part of a line is likely to stick to the nozzle. On the initial layer, small features like screw holes or tiny supports often don't stick at all.

**Describe the solution you'd like**
There should be an option for the initial layer to print features in order from large to small, or maybe simply print holes after the bottom. Additionally, any concentric brims inside a hole (including for supports) should be printed from the outside in.

**Describe alternatives you've considered**
- `Small Feature Initial Layer Speed` sometimes helps, but it doesn't work for supports.
- #7324 suggests modelling the bottom layer smooth with the screw holes to be drilled out later, but again, it doesn't help with supports.
- Increasing `Wall Line Count` might help, but it affects every layer.
- Setting `Initial Layer Support Line Distance` to the nozzle width instead of using `Enable Support Brim` works for small supports, except when also using `Cubic Subdivision` infill since it requires `Infill Overlap < 0`, which affects supports as well and causes a gap (see #9108).

**Affected users and/or printers**
Anyone who prints with PETG

**Additional context**
When generating support inside a captive nut hole with `Enable Support Brim` checked, the first support brim is printed inside out unless `Build Plate Adhesion Type` is also `Brim`. This always fails with PETG.

Inside out (fail):
[image]

Outside in (success):
[image]

Project file (same as #9108):
[hole-small.zip](https://github.com/Ultimaker/Cura/files/5795243/hole-small.zip)
Printer definition:
[bibent.zip](https://github.com/Ultimaker/Cura/files/5795246/bibent.zip)

fvrmr, 2021-01-15:
Will discus this with the team. I'll let you know.

Ghostkeeper, 2021-01-21:
From our discussion, one thing we agreed on was that these inside brims would be better to print in opposite direction. But then there is also the hybrid case, something like this:
[image]
Some of the lines on the inside are actually outside lines (the same continuous line as on the outside of the C) while some of the lines are actually like lines in a hole. So it would then start somewhere in the middle of the width of the brim? That wouldn't be very advantageous for adhesion.

JohnTasto, 2021-01-21:
Nice example. I know how I would want that to print, but I don't know how calculable it would be for the general case.

1. Outside lines
2. Hybrid lines, each beginning with the long outside arc before moving inside
3. Inside lines

Something like this might work more generally:

1. Calculate each brim line's average curvature along the line
    (something like the sum of the absolute values of each segment angle divided by total line length)
2. Print least curved lines first
3. Begin each line just before its lowest curvature

I can already think of several issues though:

- A 1mm straight line has 0 curvature, but nevertheless would be impossible to print without neighboring lines
- A line may start on a 1mm straight stretch (0 curvature), followed by a series of hairpin corners (unaccounted for)
- Needs to be balanced with travel optimization

The first two should be solvable, but this problem is more complex than I had realized. Are there any other cases to consider besides hybrid lines?

Ghostkeeper, 2021-01-25:
Well, not necessarily another case but there are a few other optimisation priorities to consider:

* You generally want all of the brim lines to print adjacent to each other. Single lines don't stick well to the build plate, so crossing single brim lines is liable to fail your print. So as much as possible, it should print the brim lines in sequence.
* Part of the purpose of the brim is to prime the nozzle. During the first few lines, the nozzle won't be properly primed yet, so it cannot reliably extrude material.

These two considerations are the reason why it's currently starting far away from the model, then closing in in perfect sequence.

If we can ensure that the outside brims are printed first, then the second issue is not a problem any more, since the nozzle is properly primed by then, and we may decide to print inner brims the other way around (but still in sequence). The hybrid example I gave before makes this difficult. We'd need to decide per group of contours in which direction to print them, so that only the non-hybrid holes in the layer get printed from close-to-far (i.e. starting near the outer wall, then going away from the outer wall).

JohnTasto, 2021-01-29:
> So as much as possible, it should print the brim lines in sequence.

I was thinking sorting by curvature would solve this, but actually printing two identical models would cause it to alternate back and forth between them, one line at a time. But if there are separate groups of contours for each and each group is printed until completion, what else could cause it to print non-adjacent lines? Outer lines should print far-to-near, inner lines should print near-to-far, and hybrid lines should choose the best option based on curvature.

I need to do some sample curvature calculations to see if a simple average could even be a useful metric. Looking again at your example, I'm not sure how much impact the sharp corner has on the outermost inner brim line - that is, I wonder how much it mitigates the larger overall radius.