Ultimaker/Cura #9111: Ultimaker 2+ firmware failed. No power

dukelazowski, 2021-01-11:
**Application version**
(The version of the application this issue occurs with.)
Ultimaker 2+ and Cure 4.8 cura update
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
I purchased 2 Ultimaker 2 for East Stroudsburg school district tech ed department. I tried to update the firmware on one of the machines. I started an update and it stopped. Received "Firmware update failed due to an unknown error." I contacted Ultimaker support, they wanted me to purchase another power cord for $165. I got the power cord from the same machine from the other school in the district, there was nothing wrong with the cord it powered the other machine. Now support wants me to buy an electronic board. I don’t think that’s the problem either. My machine was working fine before I attempted the update. Now my machine will not respond to anything. It is getting some power because the electronic board still has a green light on and the machine is making some faint noises.
First are you aware of a firmware fix?
**Printer**
(Which printer was selected in Cura?)
Ultimaker 2+
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
Contacted Ultimaker Support, they want me to purchase a $350 electronics board
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
No images. I do not have a LCD display on my Ultimaker
**Actual results**
(What happens after the above steps have been followed.)
Looking for a firmware update that will restore my machine
**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-12:
You can find this firmware here: https://github.com/Ultimaker/UM2.1-Firmware
But this is the same firmware you'll find in Cura.

jellespijker, 2021-01-18:
Hi @dukelazowski 
It almost looks like the firmware upgrade was interrupted mid-way, which resulted in a bricked UM2. This isn't technically an Cura related issue. But I might be able to help you out.

You say the power-cord you use is still working. That is great. Dit you also try it with a new USB cable?

It could very well that your board is broken due to the fact that the previous firmware upgrade was interrupted, leaving the embedded programmer, the chip that writes the firmware to the actual CPU, set in such a state that it could not recover. Since the board of the UM2  is basically an Arduino it is possible that you can flash the board using the programmer on the Arduino. But this greatly depends on your electronics and Arduino programming skills. 
If these aren't that great I strongly recommend that you don't try this and adhere to the advise of our resellers and buy a new board.

jellespijker, 2021-01-18:
Since this isn't a Cura related problem I will close this issue. But feel free to ask me for additional information.

closed 2021-01-18 by jellespijker

dukelazowski, 2021-01-20:
Yes that’s exactly what I think happened. Unfortunately I do not have any arduino experience, and only some raspberry pi. I did order a new board. Maybe I will get an arduino starter kit and try to bring the board back to life after I install the new one.

dukelazowski, 2021-01-20:
Yes that’s exactly what I think happened. Unfortunately I do not have any arduino experience, and only some raspberry pi. I did order a new board. Maybe I will get an arduino starter kit and try to bring the board back to life after I install the new one. 
Thanks for your help .

dukelazowski, 2021-02-17:
Success. Firmware installed. Back in the printing business. When I loaded the firmware to the machine it gave a pretty good surge. Actually moved the keys on the enclosure door. Probably the reason the firmware failed in the first place. Thanks for your assistance.
I did purchase an arduino to show my students how it works and tinkercad has an online component. How difficult is it to use the arduino to flash the board, since I purchased one, I’ll have an extra. 
Thanks again

Ghostkeeper, 2021-02-24:
Using the Arduino IDE to flash the Ultimaker board with Ultimaker firmware is easy! Works out of the box.

But flashing the Ultimaker firmware on a different Arduino board is probably not going to work well. It needs the stepper drivers and in the right positions and such. It can probably be made to work with a lot of fiddling but it certainly won't be easy. Not something I would feel up to the task to, in any case.

jellespijker, 2021-02-24:
Great to hear that you managed to install the firmware and are back in business again. As @Ghostkeeper mentioned flashing the firmware of Ultimaker on normal Arduino will not work well.

But you can use the regular Arduino to flash the Ultimainboard. You might look into using an Arduino as ISP https://www.arduino.cc/en/Tutorial/BuiltInExamples/ArduinoISP

jtusman, 2021-06-21:
I'm not sure why the thread keeps getting closed with no resolution.
The problem as reported in request #9111 is after installing Cura 4.9 I cannot click on any objects.
They cannot be moved or worked with at all. After completely removing Cura and reinstalling it I can use it one time.
After shut down and restart it will not work again.
It was suggested to move the Cura to my MacBook screen. This does work.
After doing testing I found if I Keep my MacBook closed Cura 4.9.1 does work on my second monitor.

fvrmr, 2021-06-22:
Hi @jtusman I think your problem has nothing to do with this thread but https://github.com/Ultimaker/Cura/issues/9912 I will mark your comment as off topic, so we keep everything in one place.

MULAWI, 2022-11-27:
> Using the Arduino IDE to flash the Ultimaker board with Ultimaker firmware is easy! Works out of the box. Just load the hex file from the resources folder of Cura (or compile your own).
> 
> But flashing the Ultimaker firmware on a different Arduino board is probably not going to work well. It needs the stepper drivers and in the right positions and such. It can probably be made to work with a lot of fiddling but it certainly won't be easy. Not something I would feel up to the task to, in any case.

Hi guys, I have a simular issue with my UM2+... I did try to upload a Firmware and it fails and a permanent soring sound comes up. Display is black and nothing goes... Cura don't recomnize my printer to try it again. I'm ona mac and how can I upload the FW via Arduino IDE now? Would be great if someone give a hand to my issue. Thanks!