Ultimaker/Cura #9116: Printer Settings Export/Import.

grk3010, 2021-01-12:
It is currently difficult to export printer settings for a custom printer definition. You can import and export materials and profiles, but not printers. 

I may take a stab at implementing this, but any pointers to what that would take would be helpful!

fvrmr, 2021-01-15:
Can you give an example for this workflow?

grk3010, 2021-01-15:
The best would be to implement it just like the "materials" or "Profiles" Import and Export so that you can export a custom printer configuration to configuration file, and then import that same printer configuration on a different computer.

For example we have 2 large format custom 3dprinters in our R&D department and 3 different users that use them. Currently each person has to manually set up a matching custom printer profile in Cura. I was actually able to manually copy the  configurations by copying the corresponding config files from the "machine_instances", "user", and "definition_changes" folders in the configuration folder.

nallath, 2021-01-15:
Cura already does this when you open a project file with that machine. 

So basicly you would do the following:
Machine with the custom machine creates an (empty) project file
Machine without the custom machine opens the project file
Select "add new" in the dialog.

fvrmr, 2021-01-18:
I'm gonna close this issue, since there is a solution.

closed 2021-01-18 by fvrmr