Ultimaker/Cura #9117: [4.8.0] Object wall infill is incorrectly printed

matusnovak, 2021-01-12:
I apologize if this has been already reported. I have searched though issues and could not find any relevant to this, unless I misunderstand the root cause of the problem.

**Application version**

4.8.0

**Platform**

Windows 10

**Printer**

Ender 5 Pro

**Reproduction steps**

1. Download `Arm_RH_Remix1.stl` from here: https://www.thingiverse.com/thing:3661405
2. Import it in Cura 4.8.0
3. Rotate the object 45 degrees clockwise so it can fit the bed.
4. Enable Brim bed adhesion.
5. Slice the model with the default wall and infill settings.
6. Go to preview mode
7. Skip to slice 6
8. Run the playback and observe how the head prints the wall that is directly opposite to the [0, 0, 0] as in the screenshot.

**Screenshot(s)**

Screenshot from Cura 4.8.0 of walls being generated. Please see *Actual results* to understand what the problem is in here.

[image]

Screenshot from Cura 4.7.1 of walls being generated serving as a comparison.

[image]

Video of the first screenshot.

https://user-images.githubusercontent.com/10331278/104349957-e890d200-5503-11eb-8abc-e4438fc054d1.mp4

**Actual results**

The problem can seen when running the playback and seeing how the wall is formed (as in the first screenshot). 

The head first goes to the right position, prints a line that is part of the inside of the wall, then returns along the outside wall to a corner. Next, the head rapidly goes to the next position, correctly prints next line of the inside of the wall, then again returns along the outside wall into a corner. (See the attached video). When the head is returning, it travels along the same path -> the outer object wall. Many times over.

When running this on a real printer (Ender 5) it **prints nothing while returning along the wall**. Additionally, because it slides on top of the outside wall, **not printing any filament**, so many times, it destroys the wall in the process. It tears down the printed material. This also can be seen in the Cura playback (see video), when the head is returning, after printing the part of the infill, it does not create any new material.

This does not happen in Cura 4.7.1 at all. I have not changed any settings regarding walls or infills. When I compared the results with Cura 4.7.1 (as shown in the second screenshot) I have kept the same settings.

**Expected results**

The wall should be printed correctly. The printer head should not slide along the outer object wall over and over again.

**Project file**

The `Arm_RH_Remix1.stl` from here: https://www.thingiverse.com/thing:3661405 as described in the "Reproduction steps" section above.

G-Code:

[CE5_Arm_RH_Remix1.zip](https://github.com/Ultimaker/Cura/files/5803796/CE5_Arm_RH_Remix1.zip)

**Log file**

None

**Additional information**

None

fvrmr, 2021-01-14:
Can you share your project file, I can't reproduce it even with the reproduction steps.

GregValiant, 2021-01-14:
I think you can reproduce it by turning Combing mode on (I'm not sure which mode).  Because the part is at a 45 degree angle changing Top Bottom Line Direction to [0,90] would make a difference as it would change their relative angle to 45 degrees to the part orientation.

matusnovak, 2021-01-14:
> Can you share your project file

Sorry, my bad. I have edited the issue body and added the project file.

> I think you can reproduce it by turning Combing mode on

Yes! That's it. I had "Combing mode" set to "Not in Skin". It was chosen for me by default. Changing that to any of "Off", "All", or "Not in Infill" fixes the problem.

Is this intended?

GregValiant, 2021-01-14:
It happens when the angle between the top bottom pattern and the edges of a part is real small.  It probably wouldn't have happened if you didn't need to rotate the part to print it.

fvrmr, 2021-01-15:
> Yes! That's it. I had "Combing mode" set to "Not in Skin". It was chosen for me by default. Changing that to any of "Off", "All", or "Not in Infill" fixes the problem.
> 
> Is this intended?

I will discuss this with the team. Let you know if I have more info on this.

matusnovak, 2021-01-15:
@fvrmr @GregValiant 

Here is a video of it in action. This does not happen if I change the setting to anything other than "Not in Skin". You can also hear how it is destroying the infill in its travel path.

https://user-images.githubusercontent.com/10331278/104788319-4fc3b600-5792-11eb-87ae-219599915482.mp4

Picture:

[image]

GregValiant, 2021-01-15:
Z-hop height at .5 would probably help also.  It should keep the nozzle above the interference points.

fvrmr, 2021-01-18:
> > Yes! That's it. I had "Combing mode" set to "Not in Skin". It was chosen for me by default. Changing that to any of "Off", "All", or "Not in Infill" fixes the problem.
> > Is this intended?
> 
> I will discuss this with the team. Let you know if I have more info on this.

This is an expected behaviour and setting. And since we can not test this on an Ender-5 we can't really test this.

matusnovak, 2021-01-20:
> This is an expected behaviour and setting.

I see. Thank you for checking that for me.

> we can not test this on an Ender-5

That's fair and understandable.

Ghostkeeper, 2021-01-25:
It's expected that it makes these movements because it starts with the closest next skin line after finishing every line. Due to the geometry, the alignment of this thin strip of skin vs. the orientation of the lines, the next closest skin line is an endpoint on the other side of the skin every time. To prevent scarring the skin, it's going to travel via the walls and infill instead.

I'd say it's unlucky that it happens to go to the other side every time. This is an optimisation problem: It chooses the next line by closest eagle-flight distance, while in reality it needs to comb to the other side with a huge detour. You have a worst-case scenario here. For the infill we've added an experiment "Infill Travel Optimization" which takes those combing detours into account. But we didn't apply this to skin. The problem with this optimisation is that it makes the optimisation an O(N⁴) problem, which grows very fast with the number of lines that need to be optimised and the size of the collision areas. Since skin tends to have many more lines than infill, we've excluded skin from it to make the setting usable.

We can't test on an Ender 5 whether the "Not in Skin" combing setting is appropriate as a default. In this case I'd certainly turn it off though.

matusnovak, 2021-01-27:
Thank you for the explanation. I suppose this ticket can be closed now.

closed 2021-01-29 by Ghostkeeper