Ultimaker/Cura #9118: Cura crash - fresh install allows one time file open then crashes on each subsequent file open

pab9776, 2021-01-13:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

pab9776, 2021-01-13:
2021-01-12 19:19:34,824 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2021-01-12 19:19:34,856 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2021-01-12 19:19:34,918 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdater
2021-01-12 19:19:34,956 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzReader
2021-01-12 19:19:34,972 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeGzWriter
2021-01-12 19:19:35,003 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeProfileReader
2021-01-12 19:19:35,041 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeReader
2021-01-12 19:19:35,056 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin GCodeWriter
2021-01-12 19:19:35,141 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ImageReader
2021-01-12 19:19:35,157 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2021-01-12 19:19:35,188 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalContainerProvider
2021-01-12 19:19:35,203 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LocalFileOutputDevice
2021-01-12 19:19:35,226 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MachineSettingsAction
2021-01-12 19:19:35,241 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ModelChecker
2021-01-12 19:19:35,273 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MonitorStage
2021-01-12 19:19:35,304 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PerObjectSettingsTool
2021-01-12 19:19:35,357 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PostProcessingPlugin
2021-01-12 19:19:35,373 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2021-01-12 19:19:35,404 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PreviewStage
2021-01-12 19:19:35,426 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RemovableDriveOutputDevice
2021-01-12 19:19:35,442 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2021-01-12 19:19:35,489 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimulationView
2021-01-12 19:19:35,504 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SliceInfoPlugin
2021-01-12 19:19:35,526 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2021-01-12 19:19:35,542 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2021-01-12 19:19:35,658 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2021-01-12 19:19:35,658 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2021-01-12 19:19:35,674 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2021-01-12 19:19:35,674 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2021-01-12 19:19:35,674 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2021-01-12 19:19:35,674 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2021-01-12 19:19:35,674 - WARNING - [MainThread] UM.Logger.logException [106]: Exception: Failed to load packages for Marketplace
2021-01-12 19:19:35,689 - WARNING - [MainThread] UM.Logger.logException [110]: NoneType: None
2021-01-12 19:19:35,689 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin Toolbox
2021-01-12 19:19:35,720 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin CameraTool
2021-01-12 19:19:35,743 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin MirrorTool
2021-01-12 19:19:35,774 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin RotateTool
2021-01-12 19:19:35,805 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin ScaleTool
2021-01-12 19:19:35,821 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SelectionTool
2021-01-12 19:19:35,843 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TranslateTool
2021-01-12 19:19:35,859 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin TrimeshReader
2021-01-12 19:19:35,890 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPReader
2021-01-12 19:19:35,928 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UFPWriter
2021-01-12 19:19:35,959 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UltimakerMachineActions
2021-01-12 19:19:36,391 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UM3NetworkPrinting
2021-01-12 19:19:36,407 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2021-01-12 19:19:36,422 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin UpdateChecker
2021-01-12 19:19:36,612 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin USBPrinting
2021-01-12 19:19:36,823 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade21to22
2021-01-12 19:19:36,904 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade22to24
2021-01-12 19:19:36,982 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade25to26
2021-01-12 19:19:37,065 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade26to27
2021-01-12 19:19:37,165 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade27to30
2021-01-12 19:19:37,244 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin VersionUpgrade30to31
[... cut ...]
2021-01-12 19:50:08,115 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectSelector.qml:134:49: Unable to assign [undefined] to bool
2021-01-12 19:50:08,117 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/resources/qml/ObjectSelector.qml:137:46: Unable to assign [undefined] to int
2021-01-12 19:50:08,800 - WARNING - [MainThread] cura.UI.PrintInformation.setBaseName [366]: Unsupported Mime Type Database file extension

fvrmr, 2021-01-14:
Does it happen with every other file or just a specific file?

pab9776, 2021-01-14:
Any file. The results are the same. I get 1 load on freshly installed Cura and the second one bombs.

closed 2021-01-14 by pab9776

Ghostkeeper, 2021-01-25:
The log file indicates nothing in this case. It ends without any mention of the second file being loaded, which shows that CPython crashed. And by the looks of that screenshot, it's not a normal Cura crash but a segfault in one of Cura's dependencies.

I wouldn't know how to start debugging this, unless we're able to reproduce this ourselves. But since none of the developers has seen this so far, it seems to be specific to your platform. Perhaps your GPUs?

One thing you could try is to disconnect any external monitors. If your monitors are rendered with different GPUs (one with your Intel chip and the other one with your NVidia card), Qt is known to crash when moving to the other screen than the one it started on. Perhaps the second object is causing a pop-up to appear. STLs shouldn't create pop-ups though; only images and 3MF files afaik.