Ultimaker/Cura #9119: cura 4.8.0 appimage won't start up in MX Linux

djquincy, 2021-01-13:
**Application version**
Cura 4.8.0 appImage

**Platform**
System:    Host: <filter> Kernel: 4.19.0-13-amd64 x86_64 bits: 64 compiler: gcc v: 8.3.0 
           parameters: BOOT_IMAGE=/boot/vmlinuz-4.19.0-13-amd64 
           root=UUID=<filter> ro quiet splash 
           init=/lib/systemd/systemd 
           Desktop: Xfce 4.14.2 tk: Gtk 3.24.5 info: xfce4-panel wm: xfwm4 dm: LightDM 1.26.0 
           Distro: MX-19.3_x64 patito feo November 11  2020 base: Debian GNU/Linux 10 (buster) 
Machine:   Type: Desktop Mobo: ASUSTeK model: P8H61-MX v: Rev X.0x serial: <filter> 
           UEFI: American Megatrends v: 0405 date: 11/03/2011 
CPU:       Topology: Dual Core model: Intel Celeron G530 bits: 64 type: MCP arch: Sandy Bridge 
           family: 6 model-id: 2A (42) stepping: 7 microcode: 2F L2 cache: 2048 KiB 
           flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 9577 
           Speed: 2059 MHz min/max: 1600/2400 MHz Core speeds (MHz): 1: 1596 2: 1596 
           Vulnerabilities: Type: itlb_multihit status: KVM: Vulnerable 
           Type: l1tf mitigation: PTE Inversion 
           Type: mds mitigation: Clear CPU buffers; SMT disabled 
           Type: meltdown mitigation: PTI 
           Type: spec_store_bypass 
           mitigation: Speculative Store Bypass disabled via prctl and seccomp 
           Type: spectre_v1 mitigation: usercopy/swapgs barriers and __user pointer sanitization 
           Type: spectre_v2 mitigation: Full generic retpoline, IBPB: conditional, IBRS_FW, 
           STIBP: disabled, RSB filling 
           Type: srbds status: Not affected 
           Type: tsx_async_abort status: Not affected 
Graphics:  Device-1: NVIDIA GF108 [GeForce GT 730] vendor: Gigabyte driver: nvidia v: 390.138 
           bus ID: 01:00.0 chip ID: 10de:0f02 
           Display: x11 server: X.Org 1.20.4 driver: nvidia 
           unloaded: fbdev,modesetting,nouveau,vesa alternate: nv resolution: 1600x900~60Hz 
           OpenGL: renderer: GeForce GT 730/PCIe/SSE2 v: 4.6.0 NVIDIA 390.138 direct render: Yes 
Audio:     Device-1: Intel 6 Series/C200 Series Family High Definition Audio vendor: ASUSTeK 
           driver: snd_hda_intel v: kernel bus ID: 00:1b.0 chip ID: 8086:1c20 
           Device-2: NVIDIA GF108 High Definition Audio vendor: Gigabyte driver: snd_hda_intel 
           v: kernel bus ID: 01:00.1 chip ID: 10de:0bea 
           Device-3: Corsair type: USB driver: hid-generic,snd-usb-audio,usbhid bus ID: 2-1.1:3 
           chip ID: 1b1c:0a14 
           Sound Server: ALSA v: k4.19.0-13-amd64 
Network:   Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet 
           vendor: ASUSTeK P8P67 and other motherboards driver: r8169 v: kernel port: d000 
           bus ID: 04:00.0 chip ID: 10ec:8168 
           IF: eth0 state: down mac: <filter> 
           Device-2: Realtek type: USB driver: rtl8821cu bus ID: 2-1.3:4 chip ID: 0bda:c811 
           serial: <filter> 
           IF: wlan0 state: up mac: <filter> 
           Device-3: Realtek RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter type: USB 
           driver: rtl8812au bus ID: 2-1.5:6 chip ID: 0bda:8812 serial: <filter> 
           IF: wlx0013eff11076 state: down mac: <filter> 
Drives:    Local Storage: total: 2.73 TiB used: 54.22 GiB (1.9%) 
           ID-1: /dev/sda vendor: Seagate model: ST3000VN007-2E4166 size: 2.73 TiB block size: 
           physical: 4096 B logical: 512 B speed: 3.0 Gb/s rotation: 5900 rpm serial: <filter> 
           rev: SC60 scheme: GPT 
Partition: ID-1: / raw size: 2.63 TiB size: 2.59 TiB (98.39%) used: 54.16 GiB (2.0%) fs: ext4 
           dev: /dev/sda6 
           ID-2: swap-1 size: 2.00 GiB used: 36.9 MiB (1.8%) fs: swap 
           swappiness: 15 (default 60) cache pressure: 100 (default) dev: /dev/sda5 
Sensors:   System Temperatures: cpu: 38.0 C mobo: N/A gpu: nvidia temp: 31 C 
           Fan Speeds (RPM): cpu: 0 gpu: nvidia fan: 33% 
Repos:     No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list 
           1: deb [arch=amd64] https://brave-browser-apt-release.s3.brave.com/ buster main
           Active apt repos in: /etc/apt/sources.list.d/debian-stable-updates.list 
           1: deb http://deb.debian.org/debian buster-updates main contrib non-free
           Active apt repos in: /etc/apt/sources.list.d/debian.list 
           1: deb http://deb.debian.org/debian buster main contrib non-free
           2: deb http://deb.debian.org/debian-security buster/updates main contrib non-free
           Active apt repos in: /etc/apt/sources.list.d/google-chrome.list 
           1: deb [arch=amd64] http://dl.google.com/linux/chrome/deb/ stable main
           Active apt repos in: /etc/apt/sources.list.d/i2p.list 
[... 93 lines cut ...]
Current thread 0x00007f8b06630740 (most recent call first):
  File "/build/4.8/build/inst/lib/python3.5/site-packages/cura/UI/CuraSplashScreen.py", line 73 in drawContents
  File "/build/4.8/build/inst/lib/python3.5/site-packages/cura/UI/CuraSplashScreen.py", line 41 in show
  File "/build/4.8/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 536 in createSplash
  File "/build/4.8/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 543 in showSplashMessage
  File "/build/4.8/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 190 in startSplashWindowPhase
  File "/build/4.8/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 474 in startSplashWindowPhase
  File "/build/4.8/build/inst/lib/python3.5/site-packages/UM/Application.py", line 239 in run
  File "/build/4.8/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 341 in run
  File "/build/4.8/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 780 in run
  File "/build/4.8/build/inst/bin/cura", line 235 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.8/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.8/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

fvrmr, 2021-01-15:
This looks familiar to this issue: https://github.com/Ultimaker/Cura/issues/7836
Maybe you find your solution here. Let me know if not.

djquincy, 2021-01-18:
I've tried the steps in that post. I probably don't know what iam doing as iam pretty new to linux but it's not working for me.

jellespijker, 2021-01-19:
Hi @djquincy could you provide the complete log file(s)

You can find them at `~/.local/share/cura/4.8/cura.log` but there might als be a  `~/.local/share/cura/stderr.log`  and `~/.local/share/cura/stdout.log`.

closed 2021-02-02 by no-response[bot]