Ultimaker/Cura #9120: Problem with a slicing models which were made in Blender

Sloynok, 2021-01-13:
Hello! I have tried to make one model in Blender but soon i have discovered that in preview mode extruder was skipping some parts(it looks like it avoids certain axes in model). Then i tried the same thing with the simplest model(cube) and it happened again.I leave a link for you to understand the problem.I would be very grateful if you could help me with that.
https://youtu.be/daKRhewZVCY

Asterchades, 2021-01-13:
Are you referring to the infill section? 3D Printing - especially FDM - is seldom done completely solid. It potentially wastes material (most stresses an object will suffer are along the edges, barring certain loads) and it just takes a long time. While you draw attention to the "print speed" field for reasons I can't figure out, just above that it states that it's using 20% Cubic infill, and a little higher again it shows 2 walls and 4 skins, and that's what your video is showing. If (for whatever reason) you actually want it to be solid than you can increase the infill to 100% or up the number of walls/skins to a ludicrous amount (999 or higher, if it lets you).

If you're referring to the way the preview seems to skip across the image, instead of following your chosen print speed, that's also to be expected. The preview simulates each extrusion over a fixed amount of time, such that every movement - regardless of how long it is (or isn't) - is shown for just as long.

Should neither of those be the problem, you might want to try filling in the issue template you deleted before typing out your post. A little more detail also wouldn't hurt.

closed 2021-01-13 by Sloynok

Sloynok, 2021-01-13:
> Are you referring to the infill section? 3D Printing - especially FDM - is seldom done completely solid. It potentially wastes material (most stresses an object will suffer are along the edges, [...]
My problem was that preview which skipping across the image. 
Yet thank you very much for help.