Ultimaker/Cura #9123: Select Face to Align to Build Plate Not Working in 4.8

intakenick, 2021-01-13:
Here is a video of the issue I'm encountering: https://www.loom.com/share/57b3de4d537043a9b98c3442473520cb

fvrmr, 2021-01-14:
I tried it myself and it works fine by me. Can you share your project file? 
Do you have it also with other objects?

fieldOfView, 2021-01-14:
I see all sorts of rendering weirdness in the video that I am not sure if it is caused by the capture or something being seriously wrong with the rendering:
[image]
The application menu is green, the rotation hoops have are rendered twice, the model selection outline is offset and the wrong color. The extruder icon in the top bar and the slice button are rendered weirdly. Basically everything that has a color is rendering wrong.

Ghostkeeper, 2021-01-25:
That would explain the issue though. The user might not be clicking in the position he's expecting to click, if the entire view is shifted vertically by 20 pixels.

intakenick, 2021-01-25:
Should I do a reinstall? Issue is still happening.

--

Nick Johnson
Intake

(914) 906-3527 [mobile]

nick@myintakepro.com
------------------------------------------------------------------------------------------------

Automating preventive care. (myintakepro.com)

NOTICE OF TRANSMISSION: This email and any files transmitted with it are
confidential and intended solely for the use of the individual or entity to
whom they are addressed. If you are not the named addressee please notify
the sender immediately and delete this e-mail from your system. If you are
not the intended recipient you are notified that disclosing, copying,
distributing or taking any action in reliance on the contents of this
information is strictly prohibited.

fieldOfView, 2021-01-25:
I don't think reinstalling will change much. But please tell us if the rendering artefacts I mentioned above are visible when you use Cura, or only visible in the video you posted?

> The application menu is green, the rotation hoops have are rendered twice, the model selection outline is offset and the wrong color. The extruder icon in the top bar and the slice button are rendered weirdly. Basically everything that has a color is rendering wrong.

If all that is visible to you while using Cura, then there is something wrong with your GPU driver. Reinstalling that could help.

intakenick, 2021-01-25:
Quick follow-up: https://www.loom.com/share/03a0f74bed5845c3881a32b758c05139
File is attached.
[Back Cover.zip](https://github.com/Ultimaker/Cura/files/5867599/Back.Cover.zip)

SecurityCop, 2021-01-28:
I'm having the same issue on my personal laptop, but I don't see any of the artifacting that you seem to have. I also don't have this issue on my work computer.

https://vimeo.com/505032872

What are your computer specs?

Ghostkeeper, 2021-02-01:
The hardest issue I have ever fixed in Cura was that the disallowed areas were rendered red on some computers, which turned out to be the default colour for something that didn't have a colour in their GPU (while on most GPUs the default was black, so we didn't notice).

We calculate the rotation to rotate the model in by making the GPU render the normal vector of the surface you're clicking on. The normal vector is the direction of the surface, which we can then rotate so that it points down (-Z) so that we know how to rotate the object such that it lays flat on the build plate.
So perhaps there is something rendering differently on your computer when it renders these normal vectors. Perhaps it's rendering them with fragment interpolation while it's by default not doing that on the computers of our developers and testers?

It'll be hard to tell unless we are able to reproduce this issue on a computer ourselves, and dig into what this render buffer shows and how it arrives at the rotation it needs to perform. Unless we can figure out what the similarities are between the users that have this issue and narrow it down that way.

intakenick, 2021-02-01:
I'm on the newest Surface Pro book

-- 
Nick Johnson
*Intake*
(914) 906-3527 [mobile]
nick@myintakepro.com
------------------------------------------------------------------------------------------------
*Automating preventive care. (myintakepro.com <https://myintakepro.com/>)*

NOTICE OF TRANSMISSION: This email and any files transmitted with it are
confidential and intended solely for the use of the individual or entity to
whom they are addressed. If you are not the named addressee please notify
the sender immediately and delete this e-mail from your system. If you are
not the intended recipient you are notified that disclosing, copying,
distributing or taking any action in reliance on the contents of this
information is strictly prohibited.

SecurityCop, 2021-02-01:
> The hardest issue I have ever fixed in Cura was that the disallowed areas were rendered red on some computers, which turned out to be the default colour for something that didn't have a colour in [...]

It sounds like it's a possibility based on intakenick's reply that it could be a GPU related issue. Both of us are using 10th gen Intel laptop processors. I'm using the integrated Iris Plus graphics, and I would assume that he is using the same based on us having the same issue. Unfortunately, I do not have a thunderbolt dock to try out an external GPU to see if that were to fix my issue.  If it helps, my exact processor is the i7-1065G7. If there is anything I can do to try and help diagnose, please let me know.

SecurityCop, 2021-02-01:
> I'm on the newest Surface Pro book
> […](#)
> -- Nick Johnson *Intake* (914) 906-3527 [mobile] nick@myintakepro.com ------------------------------------------------------------------------------------------------ *Automating preventive care. (myintakepro.com <https://myintakepro.com/>)* NOTICE OF TRANSMISSION: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to whom they are addressed. If you are not the named addressee please notify the sender immediately and delete this e-mail from your system. If you are not the intended recipient you are notified that disclosing, copying, distributing or taking any action in reliance on the contents of this information is strictly prohibited.

Can I get a confirmation on your specific build? There are a few options that are available to be selected for the Surface Book 3. Do you have an nVidia GPU or are you using the model that uses the Iris Plus graphics?

intakenick, 2021-02-02:
https://www.loom.com/share/ad401811b06f41d5881984d727738e01

intakenick, 2021-02-02:
See the link above for a solution to this issue. For anyone else encountering this issue, just add Cura.exe to your Graphics Performance Preference in Windows -> Display Settings -> Graphics Settings and then switch the GPU to High Performance. Thanks all.

SecurityCop, 2021-02-02:
> See the link above for a solution to this issue. For anyone else encountering this issue, just add Cura.exe to your Graphics Performance Preference in Windows -> Display Settings -> Graphics Settings and then switch the GPU to High Performance. Thanks all.

That's awesome, glad it's fixed for you. That confirms my thought that the Intel Iris Plus graphics has some kind of compatibility issue with this. My laptop does not have a discreet GPU, so I can't go this switch, but that's at least good to know for @Ghostkeeper.

Ghostkeeper, 2021-02-03:
Yeah. I'd prefer to keep this thread open for the time being at least.

I wonder what things change when the GPU is set to High Performance. Perhaps the lower performance is always setting the normal vectors to be interpolated regardless of what the GLSL code requires?

The link results in a Server Not Found error page for me.

summX, 2021-04-01:
That didn´t do the trick on my Win10 laptop. I switched the graphics option as advised, but my objects still reorient randomly.

3Duke3, 2021-08-13:
I encounter this on both my laptop (AMD Radion + RTX3050Ti) and PC (RTX2080). After hitting the "Select face to allign to the build plate" button, you must orbit the view  first by right clicking and moving the mouse a fraction, this makes the rotation hoops dissapear and only then can you select a face to align to the build plate.

Ghostkeeper, 2021-08-20:
That workaround indicates that the issue might be that something is not re-drawing the scene after switching tools. That's a hint for what could be done to solve this.

multimike, 2021-11-06:
Also on a i7-1065g7 with integrated Iris graphics and still experience this issue in version 4.11.
The model realigns just like in the video at the start of this issue, but picks the wrong face.

@3Duke3: The issue you are describing seems to be another one. I cannot get align face to build plate to function at all.

nallath, 2021-11-08:
It could be that this fix solves this issue: https://github.com/Ultimaker/Uranium/commit/970c94b8ce5008ca90a1cbfaeeef5470879a138f

mattpenner, 2023-07-03:
Sorry to add to an old thread, but the issue is still marked as Open. I just experienced this issue and the tip from @intakenick worked great. I saw this on v5.2.2 and v5.3.1. I am on a Lenovo ThinkPad X1 Yoga Gen 6 running Windows 10. This workaround is a life saver for those where it is applicable.

neilpstevenson, 2023-12-16:
I'm also seeing this same issue in version 5.5.0 (and all previous 5.x.x versions), which is making Cura tedious to use.
My system is a laptop with an 11th gen Intel CPU with embedded graphics, which sounds similar to others have reported. I'm running under Windows 11.

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

neilpstevenson, 2024-11-20:
I have confirmed that this now works correctly on 5.9.0, so this can be closed

closed 2024-11-20 by GregValiant