Ultimaker/Cura #9124: Arachne engine alpha - doesn't slice with voron 2.4 setting

n7484443, 2021-01-14:
**Application version**
Arachne engine alpha

**Platform**
Window 10

**Printer**
Voron 2.4

**Reproduction steps**
1. setting up to voron 2.4 normal setting.
2. I tried to slice 3Dbenchy.

**Screenshot(s)**
[image]

**Actual results**
At first, the status bar fills up, and after reaching 100%, it stops at 0% and never change.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5816484/cura.log)

fvrmr, 2021-01-15:
We have a lot of issues for Arachne, see the discussions here: https://github.com/Ultimaker/Cura/discussions/8965 
We hope to solve a lot of issues soon.

n7484443, 2021-01-15:
ah, I found it was bridge issue. Thank you

fvrmr, 2021-01-18:
Since you've found the solution, I will close this issue.
Bridge issue can be found here: https://github.com/Ultimaker/Cura/discussions/8965#discussioncomment-234502

closed 2021-01-18 by fvrmr