Ultimaker/Cura #9125: Camera angle

Bog97th, 2021-01-14:
I can not long turn the view with the pointer. I like to be able to spin the build plate view.

**Application version**
4.8.0
(The version of the application this issue occurs with.)

**Platform**
Windows 10
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Ender Pro
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-15:
Can you share your log file? 
You can find that here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Ghostkeeper, 2021-01-25:
For clarity, you need to drag the right mouse button to rotate the camera.

closed 2021-01-31 by no-response[bot]