Ultimaker/Cura #9126: Obstructive member "Ghostkeeper" needs their account cancelled.

gitcnd, 2021-01-14:
This is not a bug. It's behaving as expected. Perhaps not as you desire, but it's not expected to do anything special in this case; it's just going to produce the correct amount of walls to surround those holes as the setting instructs it to, and the skin lines are interrupted by those walls. If it didn't do that, I'd consider that a bug. So this is really a feature request.

The proposed solution will not work for the arbitrary case though. It can't work for shapes which are not convex. Perhaps clearest to see with an example (same sort of colour scheme as what you were using):
[image]

It's simple enough for a human to think of a way to stagger lines such that it'll kind of work for your shape, but an algorithm has to make it work for every shape. And in some cases there just is no solution that does not make lines hang in mid-air.

_Originally posted by @Ghostkeeper in https://github.com/Ultimaker/Cura/issues/9074#issuecomment-760292411_

fieldOfView, 2021-01-14:
Ghostkeeper is the single biggest contributor to this project. I don't see why his account should be canceled in the slightest. I would even say that doing so would be disastrous for Cura. 

The only one I see misbehaving in the thread you link to is you, unnecessarily calling into question the programming skills of someone. Feel free to disagree with him, but demanding accounts to be canceled will get you nowhere.

gitcnd, 2021-01-15:
If you mean most vocal obstructionist- I agree.  I offered to fix an obvious mistake, and they wasted time by calling this bug a feature, not accepting my offer to fix it - while also demonstrating a lack of coding skills.  This is the kind of misfit who needs to be taken of f a project: bad coders and people who alienate offers of fixing do MORE HARM THAN GOOD.

nallath, 2021-01-15:
> If you mean most vocal obstructionist- I agree. I offered to fix an obvious mistake, and they wasted time by calling this bug a feature, not accepting my offer to fix it - while also demonstrating a [...]

You need to learn how to read. He did not dismiss what you said, he tried to argue a case where what you proposed might not work. I have seen cases like that happen. Right now I'm more leaning towards preventing you from making any more of these frankly ridiculous escalations. 

You original idea has merit, your escalation _does not have it at all_. In all honesty; You are being an asshole here.

closed 2021-01-15 by nallath