Ultimaker/Cura #9127: Cura closes on its own for no reason

Fabdu57, 2021-01-15:
Hello,

Since some time, I have problem with Cura.
Cura closes on its own for no reason...

I look log file, and i see that at end :

```
2021-01-14 21:09:01,495 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2021-01-14 21:09:01,495 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2021-01-14 21:09:01,511 - INFO - [MainThread] cura.CuraApplication.closeApplication [603]: Close application
2021-01-14 21:09:01,511 - INFO - [MainThread] cura.CuraApplication.closeApplication [603]: Close application
2021-01-14 21:38:33,830 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2021-01-14 21:38:33,846 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2021-01-14 21:38:33,874 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2021-01-14 21:38:33,874 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [321]: Killing engine process
```

Can you help me ?

fvrmr, 2021-01-15:
Can you share your whole log file? 

And I need more information: your Cura version, your operation system.

Fabdu57, 2021-01-15:
It is not possible to put a attachment ?

rburema, 2021-01-15:
@Fabdu57 Unfortunately, github throws away the attachments, you'll have to attach it on the site itself.

Fabdu57, 2021-01-15:
I put it on Google drive : https://drive.google.com/file/d/1vOeLXA1jSR6TtsDqrZ7da7Ob4fcFCMbv/view?usp=sharing

fvrmr, 2021-01-18:
I still need more information, can you please fill in the form with all your information.
So we know your reproduction steps, Cura version, operation system (GPU) and project file.

Ghostkeeper, 2021-01-25:
The Cura version is 4.8 and it's happening on Windows. This can be found in the log.

Still it would be very useful to know any reproduction steps.

closed 2021-02-01 by no-response[bot]