Ultimaker/Cura #9129: cura 4.8 won't totally load on windows 8 that was upgraded to windows 10

dcracer25, 2021-01-15:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-01-18:
Can you give us more information? Please fill in the form and share the log file. You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

dcracer25, 2021-01-18:
ok cura starts to load then about half way through it freezes and does nothing. but I found if I use administrator command it will load and work.

fvrmr, 2021-01-18:
Since you've found a workaround I will close this issue.

closed 2021-01-18 by fvrmr

Ghostkeeper, 2021-01-25:
Giving Cura administrator rights was probably the original issue too. If Cura is ran with administrator rights it'll create its files with administrator rights, which then can't be modified any more if you run Cura next without administrator rights.

I suggest you look through Cura's configuration folder (Help -> Show Configuration Folder), then close Cura, and change the owner of those files to your own user account rather than the administrator. The next time you should be able to run Cura normally again.

Running applications with administrator rights is bad practice. If there'd be a virus in the application it can do serious harm pretty easily. Since Cura has a plug-in system that executes arbitrary Python code, it wouldn't be very hard to make modifications that install ransomware or something.