Ultimaker/Cura #9130: Opening saved project doesn't load profile 4.8.0

TKDesigns, 2021-01-15:
**Cura 4.8.0**
(The version of the application this issue occurs with.)

**Windows 10 Pro **
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Creality CR10s Pro**
(Which printer was selected in Cura?)

**Reproduction steps**
1. Not sure what caused it. I had opened a slt file that autoscaled to 10000% (which is a whole other issue that seems totally random) but was so big it didn't show for me to select and reduce. So I cleared the build plate and unchecked the autoscale option and re opened the stl. Now when I opened a saved project it doesn't ask about updating the profiles and just loads the stl with whatever nozzle and profile are loaded at the moment. It also loads the stl on top of whatever stl is currently open on the Buildplate without clearing it first.

**Project file**
Can't select the project file as that format isn't supported!?!?!?!? Really? Don't support your own .3mf project format?

TKDesigns, 2021-01-15:
I attempted to uninstall and reinstall 4.8.0 with a restart in between but It still had the same issue. I reinstalled 4.7.1 and it works fine. Not sure what changed to make 4.8.0 no longer load saved projects correctly.

nallath, 2021-01-16:
Could you also include the project that causes this?

TKDesigns, 2021-01-16:
> Could you also include the project that causes this?

It doesn't matter which project file I try. Even old ones won't load the profiles any more. But I also can't upload the project file (3mf) since that file format isn't supported for upload.

nallath, 2021-01-18:
You can upload them if you zip them. Could you also share the log files?

TKDesigns, 2021-01-20:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5839715/DxDiag.txt)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/5839716/cura.log.zip)

Ghostkeeper, 2021-01-25:
In the log file it seems to load your 3MF file fine:

```
2021-01-15 09:09:39,043 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1733]: Attempting to read file file:///C:/Users/toolm/Documents/3D Printer/Slicer Programs/TK Designs/Base Lighted Spin V4 PLA 08 1d 4h 48m.3mf
2021-01-15 09:09:39,084 - WARNING - [MainThread] 3MFReader.ThreeMFWorkspaceReader.preRead [360]: Ignoring instance container [creality_cr10spro_0.8] with type [variant]
2021-01-15 09:09:39,116 - WARNING - [Thread-4] 3MFReader.ThreeMFWorkspaceReader.preRead [360]: Ignoring instance container [creality_cr10spro_0.8] with type [variant]
2021-01-15 09:09:39,143 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/plugins/3MFReader/WorkspaceDialog.qml:431:5: QML QQuickItem: Binding loop detected for property "height"
2021-01-15 09:09:39,145 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///C:/Program Files/Ultimaker Cura 4.8.0/plugins/3MFReader/WorkspaceDialog.qml:431:5: QML QQuickItem: Binding loop detected for property "height"
2021-01-15 09:09:41,148 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2021-01-15 09:09:41,148 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [321]: Killing engine process
2021-01-15 09:09:41,150 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [324]: Engine process is killed. Received return code 1
2021-01-15 09:09:41,166 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [189]: Extruder [creality_base_extruder_0 #2] has already been added to this stack [Creality CR-10S Pro]
2021-01-15 09:09:41,166 - DEBUG - [MainThread] 3MFReader.ThreeMFWorkspaceReader.read [701]: Workspace loading is checking definitions...
2021-01-15 09:09:41,177 - DEBUG - [MainThread] 3MFReader.ThreeMFWorkspaceReader.read [723]: Workspace loading is checking materials...
2021-01-15 09:09:41,186 - WARNING - [MainThread] UM.Settings.InstanceContainer.setCachedValues [161]: Unable set values to be lazy loaded when values are already loaded 
2021-01-15 09:09:41,193 - INFO - [MainThread] 3MFReader.ThreeMFWorkspaceReader._processQualityChanges [829]: Loading custom profile [PLA .8N] from project file
2021-01-15 09:09:41,393 - DEBUG - [MainThread] 3MFReader.ThreeMFWorkspaceReader.read [780]: Workspace loading is notifying rest of the code of changes...
2021-01-15 09:09:41,395 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [58]: Default machine actions have been added for machine definition [creality_cr10spro], do nothing.
2021-01-15 09:09:41,400 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [189]: Extruder [creality_base_extruder_0 #2] has already been added to this stack [Creality CR-10S Pro]
2021-01-15 09:09:41,402 - INFO - [MainThread] UM.Logger.info [122]: Initializing Active Machine...
2021-01-15 09:09:41,978 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.0 seconds
2021-01-15 09:09:42,022 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.015956640243530273 seconds
2021-01-15 09:09:42,031 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [556]: Calculating normals took 0.0 seconds
2021-01-15 09:09:42,036 - DEBUG - [Thread-4] UM.FileHandler.ReadFileJob.run [71]: Loading file took 0.9 seconds
```

It resulted in 3 models, apparently (since the log mentions it's calculating the normals for 3 models).

Your zip file still doesn't create the allegedly broken project file though. Could you upload that in a .zip file? That way we may be able to see if it's actually creating broken project files, or if it's just somehow not loading them properly.

In my installation of Cura 4.8 it opens the project files I created with Cura fine.

TKDesigns, 2021-01-28:
[Compass Base V5 PLA 08 23h 18m.zip](https://github.com/Ultimaker/Cura/files/5884080/Compass.Base.V5.PLA.08.23h.18m.zip)
I have both 4.8 and 4.7.1 installed on my machine. The 4.8 doesn't load properly. The 4.7.1 does. I made a quick video showing what it does. It doesn't seem to matter what project file I open, but here is one that I've tried in the video link.

https://youtu.be/kcEDqnwdkqw

Ghostkeeper, 2021-01-28:
Hmm, that project file loads fine for me, in current master as well as in the 4.8 release.
[image]

In your video it's not showing any error messages though. It appears as if you've just set the preference "Default behavior when opening a project file" to "always import models". Can you check in your preferences what that drop-down is saying? You probably want to set it to "always ask" or my own personal preference "always open as a project".

TKDesigns, 2021-01-28:
> Hmm, that project file loads fine for me, in current master as well as in the 4.8 release.
> [image]
>
> In your video it's not showing any error messages though. It appears as if you've just set the [...]

Wouldn't uninstalling and reinstalling reset this? Or is any preferences were saved, carry over to when I reinstalled 4.7? I will check when I get home from work.

TKDesigns, 2021-01-28:
It seems that was the issue. I'm not sure why uninstalling and reinstalling didn't fix it. Or why when I reinstalled 4.7.1 it didn't carry over if there was some saved setting. Thanks for the help.

Ghostkeeper, 2021-02-04:
If you install 4.8, it should take over any settings from version 4.7 unless you clicked "yes" on the question to erase your old settings. Still a bit of a user-experience problem, that question in the installer. The default for the preference is to ask the user.

If you install 4.7 after 4.8, it will not take over any settings from 4.8. It only looks for older releases. Configuration is not backported to old releases (because these old releases can't know how the configuration is structured in newer versions).

closed 2021-02-04 by Ghostkeeper