Ultimaker/Cura #9133: Cura "solidifying" hollow models

KenLucke, 2021-01-16:
Why does this model show (correctly) that it is a single body, hollow in Fusion 360, but yet when I import  it into Cura for slicing, Cura believes it's solid and made up of 2 sub-meshes and forces infill on me unless I set infill to zero.  This happens frequently when I use the "Shell" function in Fusion 360 to hollow a body, even though nothing weird shows up visually in the .stl.

The 2 "sub-meshes" (according to Cura, once I experimentally split the model and separate them) are the actual walls and the hollow space, created using the Shell function in  F360.  However, even doing that, Cura STILL thinks that BOTH "sub-meshes" are solid and creates infill.

Oh, and I've tried Exporting to .obj, Exporting to .stl, and "Saving as" .stl from Fusion 360, all of which create the same result in Cura.

**Application version**
4.8

**Platform**
Mac OSX  10.15.7

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**

1. Load Model
2. Slice
3. Ugh

**Screenshot(s)**

1. Model in Fusion 360 with section analysis showing hollow model
[image]

2. Wire frame display from .stl export in Fusion 360, again showing hollow model
[image]

3. Error checked in Meshmixer, and transparent shader applied also showing hollow model
[image]

4. Imported into Cura, Cura confirms watertight and no errors, but states it is composed of 2 meshes
[image]

5. Model "Split" using Mesh Tools
[image]

6. Sliced separated models -  filled hollow space with infill.
7. [image]

**Actual results**
Hollow body incorrectly considered solid

**Expected results**
Hollow body should not have infill

**Project file**
https://www.dropbox.com/s/kmhiji018bmej6g/Disk%20%20Float%20v2.3mf?dl=0

**Log file**
If you need a log, let  me know,  I will clear the log and reproduce to get a clean log.

**Additional information**

KenLucke, 2021-01-16:
The only way I  was able to overcome this was to take the second mesh, set it to 0 infill, walls, top layers, and bottom layers.  Even if  I just set  the infill to Zero, it still thought that there were walls to print.

[image]

nallath, 2021-01-16:
Disable the "Union Overlapping Volumes" setting.

closed 2021-01-16 by nallath

KenLucke, 2021-01-16:
OK, that simple thing (thank you) appears to work (I also found, after a suggestion in the Fusion 360 Facebook page, that creating a pinhole from inside to outside that was smaller than the slicer dealt with solved it).

I guess my question still remains:  why does it consider it two separate models when every other program I have run it through shows it as a single, hollow, entity?

fieldOfView, 2021-01-17:
> why does it consider it two separate models

Short answer: because you had "Union Overlapping Volumes" turned on.

More involved answer: because STL files do not contain information about what is an inner wall and what is an out wall. When slicing, Cura needs to know what is "inside" the model and what is "outside". NB: The hollow part is considered outside. Cura makes cross sections for each layer, and counts from the outside in; the first wall it crosses means it is inside the model, the second means it exited the model again. A third wall means you are inside again, etc.

However, the "Union Overlapping Volumes" setting basically tells Cura to ignore all walls once it has encountered the outer wall, so everything will be solid.

Ghostkeeper, 2021-01-25:
> Cura makes cross sections for each layer, and counts from the outside in; the first wall it crosses means it is inside the model, the second means it exited the model again. A third wall means you are inside again, etc.

That's a really clear explanation and good way to look at it, really. I'd like to steal this explanation for [this settings guide article](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/missing_parts.md#non-manifold-meshes) which is currently overly technical, please!

fieldOfView, 2021-01-25:
Ofcourse, be my guest.