Ultimaker/Cura #9136: [Help need solution] Linear Advance wrong on 'combined infills' , how to solve ?

studiodyne, 2021-01-16:
Hi , Linear advance just make for only one layer height , but when 'combined infills enabled' ,   linear advance value must change , because layer height change

Impossible to made it in the printer because ' Current layer height ' is unknown for the machine
Only the slicer knows the current layer height and the adjusted flow

For ex , in 0.2 layer height , if combined infills are made in 0.4 ,  at this moment the pressure in the nozzle is so low and , applying linear pressure adjustement make bad printing , and bad overlap

Idea :   Add a gcode line with ' actual current layer height' each new layer height ,  to be scripted/replaced by scripting the slicer  and if this gcode can appear , i will make a Marlin Pullrequest to add ' FIRMWARE LINEAR ADVANCE'

But for the moment , is it possile to have a variable linear advance ? I'm new in cura , i just installed it 

Thks

fvrmr, 2021-01-18:
Can you share your project file? And maybe add some screenshots to clarify this more. 

It helps for us to fill in the form so precise as possible, so we can reproduce and find a solution faster.

studiodyne, 2021-01-19:
I have remade the topic

Ghostkeeper, 2021-01-25:
Linear advance doesn't care about line height or line width, only flow rate, right? If it needs to grow to triple the flow rate, it'll provide overpressure during that grow. But that doesn't need to depend on whether it's printing 0.4mm wide lines at 0.3mm layer height, or 1.2mm wide lines at 0.1mm layer height. The result is the same.

> Linear advance is set for 0.1mm

Provably untrue. Linear advance is not in Cura at all at the moment. I think you intended to make a feature request to implement Linear Advance and then a specific variant of it. But if you just want to add an `M900` command to your starting g-code, I think you should just add it there rather than ask us to do it for you. The start g-code field is there specifically to write down all the random stuff your printer needs to properly configure and ready itself before printing. `M900` is a pretty good example of stuff that the start g-code is intended to do.

If you want to change the default start g-code for your current printer, you can. But the developers can't test those things for you, since we don't have your printer unless you have an Ultimaker printer. So we require you to then make those changes yourself, test them, and then create a pull request to merge the correct changes into the default profiles in Cura.

closed 2021-01-25 by Ghostkeeper