Ultimaker/Cura #9138: Unable to select models on the table with the mouse pointer after entering the change model type menu

glazkovigor, 2021-01-17:
**Application version**
4.8.0

**Platform**
MAC OS Big Sur 11.1 (20C69)

**Printer**
Ender 3, Coustom printer.

**Reproduction steps**
The interface works well until you click on the button to change the model type (shown in the screenshot). If you click on the button, then it is not possible to select the model on the table with the mouse. It just is not marked. In this case, you can rotate the camera, the menu items are pressed and work correctly.

[image]

The problem is seen on any loaded models.

The problem does not depend on the program settings.
Reinstallation with clearing the program cache does not help. The problem manifests itself on version 3.7.0 and higher. It was not possible to check the younger versions, they do not start on this version of the operating system.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5826159/cura.log)

fvrmr, 2021-01-18:
Can you share your project file? I can't reproduce this right now.

glazkovigor, 2021-01-18:
> Can you share your project file? I can't reproduce this right now.
Yes, I added a file.

I found a very similar problem described by issues (#8745)
Suggested solutions in the post [https://github.com/Ultimaker/Cura/issues/8745#issuecomment-756965115] also help me, but this issues of course needs to be solved somehow ...

ehusar, 2021-01-19:
One thing to add. If I leave Cura open for an extended time period, say 3-4. hours untouched, then load a new model I have this same issue.  The model cannot be selected.  If I exit Cura and restart it, load the model, then I can select it.

fvrmr, 2021-01-19:
I'm gonna close this, because it is a duplicate of https://github.com/Ultimaker/Cura/issues/8745

closed 2021-01-19 by fvrmr