Ultimaker/Cura #9139: [4.8.0] Startup time approx 40s (restart time 21s) - seems very slow

terryspitz, 2021-01-17:
**Application version**
(The version of the application this issue occurs with.)
4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10, SSD, HP Spectre laptop

**Printer**
(Which printer was selected in Cura?)
Prusa i3 Mk2

**Reproduction steps**
1. Started cura for the first time since reboot
2. Closed it, waited a few seconds, then started it again.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
n/a

**Actual results**
(What happens after the above steps have been followed.)
Takes ages to start - all in the splash/loading progress screen.
The first time, it seems to spend significant time in 'loading machines'

**Expected results**
(What should happen after the above steps have been followed.)
Loading within maybe 5s?

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)
No project loaded.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues 
to find the log file to upload, or copy a relevant snippet from it.)

[cura.log](https://github.com/Ultimaker/Cura/files/5826233/cura.log)

note lines
"2021-01-17 14:14:17,128 - DEBUG - [MainThread] cura.CuraApplication.run [833]: Booting Cura took 63.29980945587158 seconds" and
"2021-01-17 14:14:02,055 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 37.040995597839355 seconds"

**Additional information**
(Extra information relevant to the issue.)

nallath, 2021-01-18:
On a "first" boot (eg; a complete new install of Cura), it creates cache files for the various machine definitions. It's almost like it's doing that for every boot (hence it being a lot faster the second time starting it right after the previous one).

Could you have a look if the cache files are present?

KaraokeAmerica, 2021-01-18:
I have observed this for months. Versions 4.6, 4.8 and even the new alpha. I just assumed Cura was slow to load, but I have to admit it's a bit annoying since programs like MS Office and other multimedia apps open considerably faster.

nallath, 2021-01-19:
> I have observed this for months. Versions 4.6, 4.8 and even the new alpha. I just assumed Cura was slow to load, but I have to admit it's a bit annoying since programs like MS Office and other multimedia apps open considerably faster.

I only have logs from you running 4.6 where it's super slow. (70+ seconds to load all the profiles). This should be faster with 4.8.

Could you provide me with a set of (clean) logs of a single 4.8 boot?

terryspitz, 2021-01-19:
Thanks for your reply.

What do you mean by clean logs?  I can give you the date times of the two
boots in that log if that helps.

jellespijker, 2021-01-19:
@terryspitz You can delete the log files and run Cura again from scratch, that way it will be less cluttered and easier to read for us.

Ghostkeeper, 2021-01-25:
Still, it's particularly slow for him. I don't have an SSD, just a hard disk, and for me Cura starts in just over 6 seconds. I have 7 printers added to Cura at the moment.

A log file of Cura 4.8 would be very useful. Then we can see what it's spending so much time on. The log file from 4.6 isn't all that useful any more due to the performance optimisations made in 4.7.

terryspitz, 2021-01-30:
Clean startup log attached from today, startup after reboot took >30s.

[cura_startup_20210130.log](https://github.com/Ultimaker/Cura/files/5898120/cura_startup_20210130.log)

note log file says loading metadata took 29s:

2021-01-30 13:40:33,042 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 29.157631874084473 seconds

Ghostkeeper, 2021-02-05:
Okay, so for you the loading of container metadata is 29s out of 54s, just over half of the start-up time. While in Oliof's ticket it only takes 0.5s out of 10.5s, which is more like 5%. On my computer today it takes 0.9s out of 3.0s total, though already having launched Cura earlier today the files will largely be in RAM. Interesting how widely different that can be between different systems.

Liger0, 2021-03-04:
My Cura 4.8, on an ssd, takes 40s to load. 2 Machines configured.
[cura.zip](https://github.com/Ultimaker/Cura/files/6086824/cura.zip)

MatixYo, 2021-07-09:
I have the same issue on Cura 4.10 and the developer version of Windows 11. It takes 1-2 minutes for Cura to open. Other apps open quickly.

nallath, 2021-07-12:
@MatixYo; If you want to provide us with more info, please also post log files. Just stating that it also happens for you doesn't really help us much.

MatixYo, 2021-07-12:
@nallath `[cura.log](https://github.com/Ultimaker/Cura/files/6801273/cura.log)`
Total time was about 1:30 m.
`UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 64.81982064247131 seconds`
Should that one take that long?

nallath, 2021-07-12:
It shouldn't, no. On my machine it takes 1.6 seconds to load all plugins and 1.23 seconds to load all metadata. (vs 8.2 and 64 on your system)

KaraokeAmerica, 2021-07-12:
I have been having this issue for months and in fact I commented myself in January.
[CuraLog.log](https://github.com/Ultimaker/Cura/files/6801569/CuraLog.log)
I have attached the log file I found for a fresh start today to see if it says anything useful. It is the only log file I could find. I renamed it to not confuse Cura.

MatixYo, 2021-07-12:
On Windows 10 Cura launched quickly. Now besides Windows 11 I also have WSL enabled (but all instances off).

jrwaters2, 2021-07-13:
I’m having significant delays running 4.10.0.  I erased my cura.log before doing this so my attached cura.log will be clean and only show the two startups that I did for comparison.

**I found that running Cura as an administrator is a workaround for the slow behavior.**

So I’ve run two scenarios for comparison:
First time, I ran Cura as administrator.  Startup was fast.
Second time I ran it normally and startup took minutes, stalling at “Loading UI”.

In cura.log, look at first startup, specifically these entries
2021-07-13 12:18:57,098 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [352]: Post processing view created.
2021-07-13 12:18:57,247 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///C:/Program Files/Ultimaker Cura 4.10.0/resources/qml/PrintSetupSelector/Custom/MenuButton.qml:42:18: QML Label: Cannot anchor to an item that isn't a parent or sibling.

There were 150 ms between these two entries.

The second time, [I *think*] the same sequence is:
2021-07-13 12:20:28,507 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin._createView [352]: Post processing view created.
2021-07-13 12:36:05,745 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///C:/Program Files/Ultimaker Cura 4.10.0/resources/qml/PrintSetupSelector/Custom/MenuButton.qml:42:18: QML Label: Cannot anchor to an item that isn't a parent or sibling.

**Which took 16 minutes.**

Now that I know this, I can run Cura as an administrator as a workaround.  I need to do this because it isn’t just startup that is slow, its any action involving preferences including looking at profiles or managing materials.
[cura.log](https://github.com/Ultimaker/Cura/files/6811501/cura.log)

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

terryspitz, 2024-12-05:
Cura 5.9.0 took 49s to load from a (long completed) Win11 machine reboot.
about 25s were in the first 'Loading package manager' stage.

GregValiant, 2024-12-05:
I just checked and mine was 23 seconds.  It has been longer on occasion.  In contrast Excel starts in about 2 seconds.

GregValiant, 2026-05-30:
Is this still a problem in 5.12.1 or can this be closed?