Ultimaker/Cura #9140: move part

Rick-TRIGGER, 2021-01-17:
**Application version**
(The version of the application this issue occurs with.)
CURA 4.8

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
MAC Big sur

**Printer**
(Which printer was selected in Cura?)
Creality ender 3 et cr10s Pro V2

1. je ne peux pas selectionner la piece afin de la tourner.
2. je selectione la piece par le nom du fichier et quand je tente de bouger la piece avec les fleche ,la piece se deselectione

fvrmr, 2021-01-18:
Can you explain this in English please? 
Can you share your log file and maybe some screenshots of what is happening? 
You can find your log file here: `$USER/Library/Application Support/cura/<Cura version>/cura.log`

Ghostkeeper, 2021-01-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/8745

closed 2021-01-25 by Ghostkeeper