Ultimaker/Cura #9142: More control in Monitor Tab

gaz10elle, 2021-01-18:
This is more of a suggestion for future versions. I love having my printer directly connected to my computer and find it convenient to print directly from Cura. However, many of the features available to tune the print are unavailable in the Monitor tab. I like the flexibility to Tune the print when I print using the microUSB, but would prefer those Tune abilities to be available in the Monitor tab in Cura once the print begins; much like the Tune features on the printer itself. Any chance this could happen?

fvrmr, 2021-01-18:
I'll discus this with the team. Keep you posted!

fvrmr, 2021-01-19:
I discussed it with the team and since this will not benefit for our printers, we will not look into this. But feel free to build a pull request.

closed 2021-01-19 by fvrmr